ratgdo/homekit-ratgdo #172: [feature request] Add the ability for precise door opening

ItsHaysus, 2024-05-06:
Not sure if this is the right place but when I was researching I saw that you could say… open the door halfway at 50%

Now I know Hk only allows open or closed. But adding the ability in the webpage ti open the door to a specific percentage would be a nice option.

jgstroud, 2024-05-06:
Probably won't add this, but you can do it with the eaphome firmware and Homeassistant or with the homebridge plugin

dkerr64, 2024-05-06:
I agree with @jgstroud , I don't see this in scope at all.  It is not supported by HomeKit and I believe that some calibration is required to implement (like... calculate the time it takes to open a door fully, say 20 seconds.  Now I know that to get half way I should send door open event, followed by a command to stop it 10 seconds lates).  Not something I want to get into.

jgstroud, 2024-05-06:
I see the sad face, but really, this is already supported if you use the homebridge plugin.