ratgdo/homekit-ratgdo #181: Unable to connect to Wi-Fi

K-Bert9987, 2024-05-22:
I just received ratgdo board today. This evening I tried to set it up. First I had trouble getting the driver to work. It tookna few reboots and reinstalls but it finally got there. Then I loaded the latest firmware for the MyQTT. I am planning on using it with my EISY. I recieved the succes notification and was then prompted to put in my Wi-Fi network name and password. It then tried to connect. I left it for twenty minutes without success. I double checked my router for any new devices, but there were none. I then backed out and reconnected and was given theboption to view the terminal and logs. I tried using thr reset device button o  this terminal and was able to watch it try to connect to wifi several times.

Still no success. I reloaded the firmware two more times with the same results. Any advice?

-K

dkerr64, 2024-05-22:
What firmware version are you using?  You mention MyQTT but this repository is specifically and only for the ratgdo HomeKit firmware.

jgstroud, 2024-06-03:
Closing as the original poster never responded.  please reopen if this is still an issue