ratgdo/homekit-ratgdo #187: 1.4.2 Closed status delayed by minutes

gfreestone, 2024-05-23:
Whenever door closes, HomeKit doesn't get the notification/update for several minutes.  
[image]

jgstroud, 2024-05-24:
The title says 1.4.2.  Is this a regression?  Is this consistent?  I'm not sure what can cause this.  The characteristic gets updated immediately.

jgstroud, 2024-05-24:
Interesting.  I just reproduced the issue.  The open updated immediately, but not the closed.  Hopefully an easy fix.

gfreestone, 2024-05-24:
Just tested 1.4.3 and it seems much better! Thanks!

jgstroud, 2024-05-24:
I don't know why HK seems to miss the closed state change.  It's the same code that sets the open state change and that doesn't seem to be missed.  And this was working at some point.  But I had noticed recently that the closed state was taking a while to update from my car.  I had just assumed it was delays due to remote (not wifi) updating.