ratgdo/homekit-ratgdo #189: "checkFlashCRC() failed, reboot and/or flash new firmware" after updating to 1.4.x

clajiness, 2024-05-24:
Hello,

After updating to `1.4.1`, I received the warning `checkFlashCRC() failed, reboot and/or flash new firmware`. I rolled back to `1.3.5` without issue. I tried `1.4.3`, but saw the same warning and decided to roll back again.

I did not grab logs, but would be happy to upgrade and collect them if they would help.

Thanks much!

dkerr64, 2024-05-27:
Version 1.3.5 did not report flash CRC state, it was something we added after that.  If 1.4.3 is constantly reporting failure, then that is concerning and I would advise updating firmware from USB.  If you still get the error after that, then repeat and this time select the erase option.

clajiness, 2024-05-27:
Thank you for the info. I'll give the update a try via USB and see how it goes. Appreciate it!

dkerr64, 2024-06-02:
We have found a bug that was overwriting flash memory that we shouldn't have been.  We're working on a fix.  I suspect that the "OTA bricked my ratgdo" is not quite accurate... the flash had already been corrupted and a reboot even without an OTA upgrade would have failed.

The reason that everything seemed to be fine is because the part of the flash that was written over is in the first 4KB which is reserved for the bootloader... and of course it only ever runs at boot.  But it was in the HomeKit server, so it was writing (actually erasing) a part of memory that it shouldn't... and the part that should have been erased was not.  This likely led to problems with HomeKit server operations.

Time will tell if we have finally found the root cause of our stability problems.

jgstroud, 2024-06-03:
Fixed in 1.5.0.  please reopen if you see this again.

closed 2024-06-03 by jgstroud