ratgdo/homekit-ratgdo #192: Changing Wifi Tx Power - Soft Bricks on 1.4.3

zacshann, 2024-05-29:
Hi,

So at first I thought this was a fluke, but I was able to reproduce this on two separate ratgdo 2.53i's both on 1.4.3.

I adjusted the Wifi TX power (which I have done numerous times before on lower firmware) from 5dbm to 8dbm (but just going up seems to do it, I was able to change from 3-5 and cause the same thing on the second unit), and the board rebooted and never came back. You will actually notice on the physical board the normally always on blue LED will go out when this occurs. Removing power from the board & reapplying will cause the blue LED to flash for less than a second and then go off again, the board will never reconnect to Wifi or revert its TX power in this state. 

The above being said, reconnecting the unit to a computer, and re-flashing the firmware resolves this, and I was also able to adjust the WiFi Tx power down from the default of 20 dbm to 8 dbm after re-flashing without issue. (I didn't try adjusting up again after this point as I didn't want to climb back up  to my gdo's & reflash the door again/ re-set the units up in homekit)

I tried to see if logs were on the unit prior to re-flashing but they appeared blank when connected to the firmware/flashing portal, it also doesn't appear to be an overall TX power issue, as the units are sitting next to an AP when this occurs.

jgstroud, 2024-05-30:
I'm not able to reproduce this error, but it seems that when the setting is written to flash it must be corrupting some setting.  I have implemented a change that should greatly improve this in the next build.

jgstroud, 2024-06-03:
I believe this problem should be fixed now.  Please reopen if this remains an issue.

closed 2024-06-03 by jgstroud