ratgdo/homekit-ratgdo #194: Terminal Loss of Wifi Connection

blueyetisoftware-jason, 2024-06-03:
I have 2 devices that were each running a prior firmware. I cannot comment on which one because I was unable to connect to them and a direct connection yielded a log of 0 bytes. Over the past month, they each went offline and did not return.

I flashed these devices to 1.4.4 without erasing the devices and it installed successfully. The device functioned correctly while remaining connected directly to the computer. I unplugged the device and plugged it back into a usb adapter. At that point, the device would not come back online. Restart attempts were futile.

I repeated this with both devices and was unable to get them to survive the disconnect from the computer after flashing them.

I temporarily switched these devices over to the ESPHome firmware instead to get them functioning and make sure I didn't have a power problem. The ESPHome firmware is working as expected and reconnects on reboot. I did erase the device when switching to ESPHome, so it is possible the "erase" may have fixed the issue as well.

This feels like something is being corrupted on a loss of power.

jgstroud, 2024-06-03:
Please try 1.5.0.  There was a bug with how things were being stored in flash.

jgstroud, 2024-06-03:
Closing as I believe this issue is corrected in version 1.5.0.  Please reopen if it continues to be an issue

closed 2024-06-03 by jgstroud

blueyetisoftware-jason, 2024-06-03:
Thanks for the quick feedback. For what it's worth I like this driver better than ESPHome. The ESP route opens up more controls in HomeAssistant, but it also has its own bugs. I tried the remote lockout over there and it completely messed up my openers. It locked them and they wouldn't unlock, even from the wall panels. I had to pull the power to the GDO itself together the wall panels to reset.

Looking forward to switching back to this version.

jgstroud, 2024-06-03:
The esphome firmware and plugins definitely have more features.  use the one you like best or works best for you.  hopefully we fixed the issue you reported, but its hard to say 100% for sure unless you try and let me know