ratgdo/homekit-ratgdo #195: cannot re-pair 2 ratgdo devices with fw 1.5.0

OleSam85, 2024-06-03:
I upgraded both my ratgdo devices to 1.5.0 and cannot pair both of them. They have the same pairing code. I was able to pair one by rebooting the other one and starting the pairing process on the first while the other was down. Now though both devices (the paired and un-paired one) show that they're paired with homekit. If I un-pair the one that's not paired, i suspect it will somehow break the connection with the one that is paired.

jgstroud, 2024-06-03:
The most reliable way to do this is 1) erase from Apple Home, 2) reset/re-pair button on ratgdo, 3) kill Apple Home and restart it and 4) scan the QR code.

jgstroud, 2024-06-03:
I recently paired 2 devices using this procedure.

OleSam85, 2024-06-03:
Yes that seemed to work, the order of operations is critical apparently; first time I tried I think I killed Apple Home app too soon and when I tried to add it I got an "Accessory Already in Another Home" error. Anyway I killed the Apple Home app one more time and now I seem to be ok. Thanks for the quick reply.

closed 2024-06-03 by OleSam85