ratgdo/homekit-ratgdo #198: HomeKit garage door not working after 1.5 update

bob3k3, 2024-06-05:
After 1.5 update, HomeKit would not connect to door and icon will not control either door or light.. I deleted the HomeKit item and then tried to associate the garage door again with my HomeKit. I got a message that the garage door was associated with another home. And then tried to reinstall the software and I also get a message."WARNING: Flash CRC check failed. You must flash new firmware by USB cable to recover, please consult documentation. RATGDO device may not restart if you reboot now. Reboot anyway?"  

Not sure if this issue is something that I did wrong. Probably.

jgstroud, 2024-06-05:
Probably less that you did something wrong and more that the firmware prior to 1.5 had a pretty fatal bug.  Since you have the CRC error, you probably need to reflash from USB.  As @dkerr64 put it:
I have found that the most reliable way to do this is 1) erase from Apple Home, 2) reset/re-pair button on ratgdo, 3) kill Apple Home and restart it and 4) scan the QR code.

bob3k3, 2024-06-06:
Thanks, that’s exactly what I did and everything’s working fine. Love the Ratgdo! Best, Bob

closed 2024-06-06 by jgstroud