ratgdo/homekit-ratgdo #209: Unable to connect to WiFi. May be unicode issue?

marcushasfun, 2024-06-29:
My WiFi's SSID is "It's an 802.11g thang". As you can see from the attached screenshot ratgdo is having trouble with that apostrophe. There are also special characters in the password. 

I don't know if this is why ratgdo is failing to connect but I have tried multiple times and it is failing to connect.

I have an Eero Pro 6E network.

[image]

marcushasfun, 2024-06-29:
This is what i am seeing in the logs -

>>> [  61233] RATGDO: WiFi disconnected SSID: Itâs an 802.11g thang, BSSID: 00:00:00:00:00:00, Reason: 201

jgstroud, 2024-06-29:
So, I think it's not handling the SSID name properly.  Reason 201 is AP not found.  I can't look at the code now, but If you are comfortable working from the command line, there is an improv.py script in the git repository.  You can try using that to connect to the WiFi and see if you get any further.  Usage is like:

./improv.py -d /dev/ttyUSBX -s "It\'s an 802.11g thang" -p "my password"

You'll need to modify the device and you may need to experiment with how exactly to pass your unique SSID.

marcushasfun, 2024-06-29:
> So, I think it's not handling the SSID name properly. 

That's what I figured. Unfortunately I have many IOT devices that would have to be reconfigured if I change the SSID name.

I'm OK with the command line but it sounds like this is a bit more than that. It's not urgent so I'll wait until you can have a look at the code.

Much appreciated.

jgstroud, 2024-06-30:
I'm not sure what character you have in your SSID.  I copied and pasted from your message and created a new SSID, but it shows up fine and I am able to connect.  Can you tell me what special characters are in your password?  I put an ! and a @ in my password and had no issues connecting to this SSID.
[image]

jgstroud, 2024-06-30:
You might also try the "Join Other..." Option and see if you get any further:
[image]

marcushasfun, 2024-06-30:
Hmm... strange because the SSID always shows up with the apostrophe replaced by junk characters for me.

I tried "Join other..." and the logs show that the SSID is being passed correctly but it still fails to connect:

>>> [  13509] RATGDO: WiFi disconnected SSID: It's an 802.11g thang, BSSID: 00:00:00:00:00:00, Reason: 201

Could the Eero be the problem?

dkerr64, 2024-06-30:
This feels like a character encoding problem… like URL strings need certain special characters (including spaces) encoded if used in a URL.  So I would research the rules for SSIDs.

jgstroud, 2024-06-30:
Since it worked perfectly for me with what appears to be the same name, I'm thinking it's not a standard apostrophe and that it has some fancy Unicode apostrophe in the actual name.  Maybe try copy / pasting the value directly from the eero settings.

marcushasfun, 2024-06-30:
Holy guacamole! Copying and pasting the SSID name from the Eero app worked. I have no idea why typing an ' in the ratgdo web interface would be different than typing one into the Eero app, but apparently it is. Something odd is going on there.

Thank you for your help.

jgstroud, 2024-07-01:
Glad you got it to work.  I think your apostrophe is probably Unicode 2019 instead of the normal 27. 

https://www.cl.cam.ac.uk/~mgk25/ucs/quotes.html

jgstroud, 2024-07-01:
I'm going to close this issue since you were able to get connected.  This is a bug, but such a corner case that I don't have any time to spend trying to fix it.

closed 2024-07-01 by jgstroud

marcushasfun, 2024-07-01:
Fair enough. Clearly when you type an apostrophe on macOS or iOS the unicode version is used.