ratgdo/homekit-ratgdo #210: Having Issues with Ratgdo (HomeKit)

Farrissf, 2024-07-01:
I just got my RATDGO 2.53i and installed it (HOMEKIT).  It has worked flawlessly since April.  I had to cut the power to the garage this past weekend and it would not reconnect.  I pulled it off the garage door opener and hooked it to my computer and it would not connect to WiFi.  I ended up flashing it an installing 1.6.0 and after multiple attempts, I was able to get it connected to my WiFi and I installed it back on the door 
opener.  It is still not working but I think it may be cause I didn't wire it correctly.  I paired with HomeKit and you can see from the picture it appears to be okay.  I went through all the documentation and I cannot find the wiring diagram or picture I used previously.  Can you send me this information?

This is my console and a pic of my opener:

[image]

[image]

Thanks,
Scott Farris
Scottfarris@scottfarris.com
281-830-1056 Text & Cell

Farrissf, 2024-07-01:
Thank you for your response.

I just tried your suggestion, you didn't say to reboot after I made the changes but I did, First I changed it, rebooted, then I changed it back and rebooted.  Still no luck.

I am still leaning towards the wiring.  The picture I posted earlier show before the Ratdgo was installed so I went out and snapped another and I have posted it below.  Just to be clear, I attached the red wire on the far left (red) terminal and the black wire to the far right (black) terminal; the white wire which I circled is not connected to anything.

[image]

Scott

Farrissf, 2024-07-01:
Just in case you need too see this:

[image]

jgstroud, 2024-07-01:
Needs to be connected per the wiring diagram on the ratgdo page.  You definitely need the white wire connected.

[image]

Farrissf, 2024-07-02:
That worked and thank you so much!  I did have another quick question, is it a good idea, in the settings, to have the Ratdgo to restart and if so, how often.

dkerr64, 2024-07-02:
No. Don’t set a restart interval.  We added that when the firmware had some stability problems.  But those seem to be all resolved now, so I anticipate we will remove that option soon.

Farrissf, 2024-07-02:
Thank you, I am up and running again.  I have really enjoyed the RATDGO, it is so responsive. Best money I have spent in a long while.  

SFF

closed 2024-07-03 by jgstroud