ratgdo/homekit-ratgdo #211: Crash but recovered on 1.6

rdeutsch3, 2024-07-04:
No concrete information on "why" but it does potentially roughly align with my router mysteriously rebooting.

Flash CRC OK
Firmware Version: 1.6.0

ACKET(0x112918 @ 0x1DA2) Motion - NoData: [Zero: 0x00000000, Parity: 0xF]
>>> [1150665555] RATGDO: Motion Detected
>>> [1150665560] RATGDO: ENCODING 00000937 0000000000826539 00000080
>>> [1150665659] RATGDO: reader completed packet
>>> [1150665659] RATGDO: DECODED  00008C2F 00000000BE9600D2 4260B281
>>> [1150665660] RATGDO: PACKET(0x9600D2 @ 0x8C2F) Status - Status: [DoorState Closed, Parity 0xB, Obs 1, Lock 0, Light 1]
>>> [1150665669] RATGDO: tgt 1 curr 1
>>> [1150670559] RATGDO: Motion Cleared
>>> [1150670559] HomeKit: [Client 1073698068] Got characteristic 1.21 change event
>>> [1150670560] RATGDO: SSE send to client 192.168.1.166 on channel 0, data: { "garageMotion": false, "upTime": 1150670560 }
>>> [1150699741] HomeKit: [Client 1073698068] Get Characteristics
>>> [1150761644] HomeKit: [Client 1073698068] Get Characteristics
>>> [1150944407] HomeKit: wifiClient creation error, IP address is not set
>>> [1150944410] HomeKit: [Client 1073698068] Get Characteristics
!!! [1150944411] HomeKit: [Client 1073698068] The socket is null! (or is closed)
!!! [1150944417] HomeKit: [Client 1073698068] The socket is null! (or is closed)
!!! [1150944423] HomeKit: [Client 1073698068] The socket is null! (or is closed)
>>> [1151116295] HomeKit: wifiClient creation error, IP address is not set
>>> [1151116296] HomeKit: [Client 1073698068] Disconnected!
>>> [1151116296] HomeKit: [Client 1073698068] Closing client connection
>>> [1151288034] HomeKit: wifiClient creation error, IP address is not set
>>> [1151459898] HomeKit: wifiClient creation error, IP address is not set
>>> [1151631737] HomeKit: wifiClient creation error, IP address is not set
>>> [1151803502] HomeKit: wifiClient creation error, IP address is not set
>>> [1151975590] HomeKit: wifiClient creation error, IP address is not set
>>> [1152147454] HomeKit: wifiClient creation error, IP address is not set
>>> [1152319418] HomeKit: wifiClient creation error, IP address is not set
>>> [1152491084] HomeKit: wifiClient creation error, IP address is not set

dkerr64, 2024-07-04:
We have seen crashes when the WiFi network goes out, usually in the mDNS code.  Did you clip out some of the stack dump?  It looks incomplete.  If you can attach the full crash log then we may be able to decode the stack dump and see if it is in mDNS. 

The good news is that it should recover just fine. And you should delete the crash log (after you save a copy) so that there is room to save any future crashes.

thanks.

rdeutsch3, 2024-07-05:
> We have seen crashes when the WiFi network goes out, usually in the mDNS code. Did you clip out some of the stack dump? It looks incomplete. If you can attach the full crash log then we may be able [...]

Yes, I truncated the sttack as from my previous life we rarely looked at them and knew where the crash was from the messages.  It was only us 'old hands' that even knew how to do it.

I've updated initial post with the entire stack, which it reports as incomplete.

jgstroud, 2024-07-05:
The stack trace is actually a lot more helpful than the log messages.  I'll try to analyze it tomorrow.  No the device does not do any communication with the outside world

jgstroud, 2024-07-05:
Yep, it crashed in the MDNS Responder.  We have seen this a number of times.  It seems to happen during or immediately after a wifi outage.  Seems to always recover.  No root cause or solution yet.  Thanks for the logs.  Hopefully we'll figure it out at some point.

dkerr64, 2024-10-19:
Possible fix in PR #242 by using IRAM heap to free up more memory.

dkerr64, 2024-10-26:
Believe fixed in 1.8.0

closed 2024-10-26 by dkerr64

rdeutsch3, 2024-11-18:
So far so good.  Thank you for the follow up