ratgdo/homekit-ratgdo #217: Not connecting to Wifi

seanpopke, 2024-07-15:
I ordered two ratgdos last fall. I installed both and both were working correctly. I recently noticed that neither were showing up in Apple HomeKit (I had installed the native HomeKit firmware.) When trying to reboot both, the blue light flashed once, which seems to indicate that they had lost their firmware? 

I’ve taken one into my office and reflashed it with the latest HomeKit firmware (1.6). The flash seems to work, but when I try to connect to Wifi, it isn’t successful. The logs show:

[58915119] RATGDO: WiFi disconnected SSID: iot24, BSSID: 6a:d7:9a:24:fa:6a, Reason: 203

I’ve tried erasing before re-flashing, but no change. Nothing has changed on my network or wifi access points. I’m using a Unifi system with a dedicated 2.4Ghz IOT SSID.

Any thoughts?

dkerr64, 2024-07-15:
The reason code 203 maps to WIFI_DISCONNECT_REASON_ASSOC_FAIL.  The ratgdo should continue to try and reconnect.

 I suggest looking at your Unifi logs to see if it gives any clues.

jgstroud, 2024-07-15:
Do you have them locked to an AP?  A lot of users did this early on, but it seems to be causing issues lately.  I recommend not locking the devices.

jgstroud, 2024-07-15:
Also, did you enable PMF on your wifi network?

seanpopke, 2024-07-15:
Thanks for the responses so far. They are not locked to an AP and PMF is disabled. I've made no changes to the wireless network between when I first set them up and now.

marylegz90, 2024-07-17:
I, too, am having a similar issue. I've updated my 2.5 to 1.6.0 after finding it "No Response" in HomeKit. Took a bit and a few reboots, but finally got the firmware web installer to get it "connected" to wifi. I found that it still is unresponsive in HomeKit but does show to be up in my Unifi console (Dream Machine Pro SE running 4.0.6). It doesn't stay connected hardly at all though, very randomly will it even ping.

PMF and AP locking is turned off on my network. No multicast or IGMP snooping is enabled either - as I had seen someone else say that worked for them.
[image]

Logs every once and a while throws back "Reason: 4" followed by multiple "Reason:203s" before cycling back through.
[image]

Latency doesn't even report on the device sitting a few feet away from a U6-LR AP that is running a dedicated 2.4ghz network.
[image]

jgstroud, 2024-07-17:
Just searching on the association fail, there seems to be a lot of results but no answers that really point me in the right direction.  Without being able to reproduce, I'm at a loss.  Anything at all in your unifi logs? If you can get it to connect at all to wifi, you can try forcing 802.12g or try reducing your wifi power level.  One comment said dropping to 8.5 helped.  https://community.home-assistant.io/t/unable-to-connect-to-wifi-auth-expired-and-association-expired/678570/3

marylegz90, 2024-07-17:
> Just searching on the association fail, there seems to be a lot of results but no answers that really point me in the right direction. Without being able to reproduce, I'm at a loss. Anything at all [...]

Thanks for the reply. It ended up randomly connecting at 1:35am (I was asleep), so not sure what caused it to make the connection. Since then it dropped out twice on it's own. I got home from work and it was connected, so I ended up going in and doing the following: lowered the RAGTDO's Wifi TX Power to 8, reducing my AP's 2ghz transmit power to low, rebooted the RAGTDO and it successfully connected.

Monitoring it in my Unifi controller I see lots of TX Retries, from this device only. It bounces from 0% to 75% and in-between often.
[image]

So far, it's at least staying connected and I can see it in my Homekit app now, even controlled the door and light successfully. I'll see it stays up for an extended period of time.

jgstroud, 2024-07-18:
That says "WiFi 1".  Did you also change the wifi standard to 802.11b?

marylegz90, 2024-07-18:
> That says "WiFi 1". Did you also change the wifi standard to 802.11b?

I did not, it's setup for auto, still.
[image]

jgstroud, 2024-07-18:
Progress I suppose.  I also use auto mode and Unify reports my device as WiFi 4.  I am by no means any kind of wifi or RF expert, but the fact that you had to dial down the power and its showing up as WiFi 1 makes me think you are having some sort of interference.  How does your channel scan look?  Is channel 11 crowded?  Do you have anything in or near your garage that might be causing interference on 2.4G?

I see you have an LR access point.  I can't say I really know what makes these devices LongRarnge other than maybe upping the TX power on them.  I have some of the AC-LRs in my office and had the turn the power down on them.  You said you turned yours down as well and I think that may actually be what fixed it.  You may just be over-saturating the receiver on the ratgdo.  I don't think they have the greatest DSP's in the world and may be struggling recovering the signal.  

That's probably a whole lot of not very helpful information.

dcrousso, 2024-11-14:
so ive tried having it reboot every 24hrs and 1hr and etc. but none of those usually work

but i have noticed that actually physically disconnecting the RATGDO from power seems to allow it to reconnect to WiFi

perhaps there's some residual power even in the case of a reboot that's somehow not playing nicely with my router?

megamuffin13, 2024-11-14:
TP-Link Archer AX11000 also has a Smart Connect feature, you might toggle it on or off in your router settings and see if that helps.

https://www.tp-link.com/us/support/faq/2595/

megamuffin13, 2024-11-15:
Conversely you can create Separate WiFi SSIDs and passwords for 2.4GHz and 5GHz Bands, so that the RatGDO only accesses the 2.4 ghz ssid.

dkerr64, 2024-11-15:
ratgdo (the ESP8266 chip it is based on) can only access 2.4Ghz so will never attempt to connect to 5Ghz.  I think it highly likely that the WiFi stack on this will not support some of the newer optimization techniques used by WiFi access points.  I don't know what Smart Connect does, but try disabling this, or similar.

I do think it is a good idea to create a separate VLAN and SSID for smart devices in general, so definitely try that.

If you have multiple WiFi access points, we do now have a way to bind ratgdo to a specific AP, which may help.

Unfortunately, other that offering suggestions, there is little we can do.  We're using all the WiFi disconnect/reconnect logic we can.  On that topic, there is no difference between a power-down/power-up and a reboot... other than perhaps the time that the device is disconnected from wifi.  Our complete reboot cycle is under 15 seconds now.

tungtt0806, 2024-11-23:
[image]
I already installed newest version but It still kepping connected and disconnected.

dkerr64, 2024-11-24:
@tungtt0806 the device is not connecting to your access point (forgive me for stating the obvious!!!) but I don't think there is anything in our code causing this.  It looks like DHCP is not issuing an IP address.  And reason 8 and 2 are...

Reason 8: WIFI_DISCONNECT_REASON_ASSOC_LEAVE
Reason 2: WIFI_DISCONNECT_REASON_AUTH_EXPIRE

Here is what Google AI has to say about those.

You might try a different WiFi network and make sure not to have anything in your network configuration that is trying to force the device to attach to a specific access point.

[image]

and 

[image]