ratgdo/homekit-ratgdo #221: Make LED polarity configurable

jgstroud, 2024-08-07:
Some users really don't like having the LED on all the time.  I personally like seeing it on to indicate it's alive and well.  We can make the polarity configurable and optionally only allow it to be on when motion is detected in the garage.  The LED may also attract bugs in some environments.

jgstroud, 2024-08-08:
Well, it already blinks on activity on the serial link.  I think we leave that part as is.

dkerr64, 2024-08-08:
So, you propose an option that essentially inverts current behavior... so if users prefer the light to be mostly off they can have that.

jgstroud, 2024-08-08:
Yeah, I think that's the easiest solution without overcomplicating anything.  But I don't really have a strong opinion about it.  Ever since it was implemented, it seems like we get about 50/50 feedback on users that would prefer it mostly on vs those that would prefer it mostly off.

dkerr64, 2024-08-08:
I've figured out how to implement so I will code it up tonight and see how well it works.

dkerr64, 2024-08-08:
Implemented in PR #220

dkerr64, 2024-08-10:
Implemented in v1.7

closed 2024-08-10 by dkerr64