ratgdo/homekit-ratgdo #222: HomeKit basic connectivity issue (maybe Ubiquiti/Unifi related)

DanaLacoste, 2024-08-10:
TL;DR : ratgdo hardware and web UI work with 100% success, but HomeKit integration fails with 0% success.

OK, I have tried many times to figure this out with no success yet.  It’s been rather frustrating.

First off: the hardware is working 100% fine.  The website comes up and I can control the light, garage door, and the motion sensor and obstruction sensor both work great.

I can also upgrade (i.e. from 1.6.0 to 1.6.1 was transparent, right through the web UI, flawless)

I have tried this on my primary WiFi and on my separate IoT SSID, in both cases I get this result:

Add device to home scans the QR code fine but can’t see the device.

I have also tried using `dns-sd` and the `Discovery - DNS-SD Browser` app (with some success: they showed my LG TV that I had not connected to HomeKit for example!) but they simply don’t see the ratgdo at all.

So I am trying to make a PR for this repo with my documentation / notes, as a guide for anyone else who gets stuck like I am, but I am really stumped as to what to do.  I have the technical background but I honestly don’t know what to do with “responds to TCP but not to mdns” : it’s like “the black box isn’t working”

Heck, I would plug it into ethernet and try that if it had an ethernet port :)

Does anyone have any suggestions for how I could “trigger” the mdns response, in some kind of testing mode?  Generate a packet and send it to the device directly maybe?

Note that I am somewhat convinced (with no evidence) that this is due to the Ubiquiti configuration I am running.  I have IGMP snooping enabled as well as mDNS and my TV is working (on same separate subnet but on ethernet, not wifi).  So I am more than happy to blame Ubiquiti, but I still want to figure it out.

dkerr64, 2024-08-10:
Is your ratgdo connected to the same IP subnet as the rest of your network.  Try the [Discovery](https://apps.apple.com/us/app/discovery-dns-sd-browser/id1381004916?mt=12) app and expand the `_hap._tcp.` section and confirm that the garage door is visible there.

If you are on a separate subnet then you need to bridge the two together at the gateway so that mDNS is sent from one network to the other... how this is done depends on your network gateway.  And if you have a firewall or packet filtering between the networks then it needs to be setup to permit traffic.

dkerr64, 2024-08-10:
When I say bridge together, what I really mean is that you need an mDNS reflector.  Personally I use avahi, but google also turns up this... https://github.com/vfreex/mdns-reflector

DanaLacoste, 2024-08-10:
I get the same result from both same network and separate network: the ratgdo never shows up under `_hap._tcp`

I have never seen it work so I don't know what I am looking for, but the LG TV is there on the list, clear as day

(Same result from command line `dns-sd -Z _hap._tcp local.` as in the app)

DanaLacoste, 2024-08-10:
WOW literally as I type that I try again and it shows up on my Mac!
```
_hap._tcp                                       PTR     Garage\032Door._hap._tcp
Garage\032Door._hap._tcp                        SRV     0 0 5556 Garage\032Door.local. ; Replace with unicast FQDN of target host
Garage\032Door._hap._tcp                        TXT     "c#=1" "sf=1" "sh=DCKuKw==" "ci=4" "ff=0" "s#=1" "id=C5:CE:C2:0A:11:29" "pv=1.0" "md=ratgdo"
```

Now to try to add it to my home....

DanaLacoste, 2024-08-10:
....and it's added but shows "No Response" in the Home app.

I am still convinced it's a network thing but will continue tinkering until I can make something document-able

dkerr64, 2024-08-10:
I suggest you try connecting the ratgdo to the same SSID / subnet as your HomeKit hub / iPhone / etc and check it works there.  That would isolate the problem to being on a different subnet. 

FYI that I also have all my IoT devices including ratgdo on a different subnet and it is working fine.

DanaLacoste, 2024-09-17:
OK, sorry this took so long, but will make a PR for documentation with my results.

Summary of behaviors observed:

- `_hap.tcp` (bonjour discovery, via command line or `Discovery` app) works sometimes, other times not
- SOMETIMES you get through the "Add to home" stage but not always
- When you view the door opener in HomeKit, it just says that it can't connect: never lets you even try to open/close

and the kicker:
- All through the frustration, the web page works 100% of the time and no reason for why homekit is failing is available

If this is you and you are using a Ubiquiti/Unifi network, try these steps:
1. Install the ratgdo on your 'home' network.  Not on a separate IoT VPN (Might figure this out in the future but for now....)
2. Disable the IoT "helper" stuff in Unifi.
  - in "MultiCast Settings", disable `IoT Auto Discovery` (which does mDNS "helping" across networks)
  - in  "MultiCast Settings", disable `Multicast Filtering` which does "IGMP Snooping" and can cause issues
3. It should "just work" now

If you figure out settings which work for "IoT device on separate subnet from home network using pure Unifi settings" please feel free to share :)

dkerr64, 2024-09-18:
Running RATGDO on a separate IoT VLAN will work... assuming that the mDNS traffic is "reflected" across both network subnets.  I have Unifi WiFi access points, but do not use their gateway.  I have my own gateway on which I run avahi and that can be configured to reflect mDNS traffic across two networks.  It sounds like the Unifi gateway is not reflecting mDNS traffic across LANs properly.

DanaLacoste, 2024-09-18:
Yeah, there is _something_ not right there, but it's non-trivial to debug.

I have a feeling that me switching the device back and forth from one wifi to another didn't help: I bet the mDNS thing was caching the wrong value or something.

So now it works (but the 'fix' broke bonjour to a printer: no problem, it's a static IP so just moved printer config to static)

Trying to come up with some clear docs on it so the next person who searches might find some useful info, and maybe eventually we get a working config going (without needing avahi) but for now _I_ am doing fine at least :)

jgstroud, 2024-09-19:
I run Unify APs and gateway / router.  It has an mdns reflector built in.  I also run on an isolated VLAN and it works just fine.  I also have IoT auto discovery enabled on all VLANs and Multicast filtering only on my IoT VLAN.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64