ratgdo/homekit-ratgdo #225: Allow setting different username

donavanbecker, 2024-08-14:
would be nice if the username could be changed from admin to something else.

dkerr64, 2024-08-17:
Request noted.  Yes username could be changed, just a small matter of programming it.  Will consider for a future release.

donavanbecker, 2024-08-17:
With that would multi user support be useful?

dkerr64, 2024-08-17:
> With that would multi user support be useful?

I think that is a stretch.  Much more complex to implement, and what's the point?  I don't want to encourage use of the web interface... it should be for admin purposes only.  Apple Home or other HomeKit apps is how most should interact with the door, not the web page.

I can think of a (quite niche case) where changing from "admin" to something else may be useful, but I'm not seeing it for multiuser.

David

donavanbecker, 2024-08-17:
Totally understand, I don't need it but figured I would propose it.

dkerr64, 2024-08-17:
Will implement in next release whenever we do that... is in PR #227

donavanbecker, 2024-08-17:
Working awesome @dkerr64!!

Anyway to change the login pop up so that it supports autofill on mobile?

[image]

dkerr64, 2024-08-17:
We're using basic browser-based digest authentication.  The popup is generated by the browser so, AFAIK, I cannot influence how that popup appears or whether the mobile (or desktop) browser will offer to autofill the fields.

The alternative would be to implement our own login dialog... that requires we implement our own and not rely on browser-based digest authentication.  This, in-fact, is best practice which is why you rarely see browser popups for userid/password anymore on web sites (it's considered insecure).  But a lot of work I think and I'm not sure it is worth the effort especially given we cannot implement SSL (https).

dkerr64, 2024-09-23:
included in v1.7.1

closed 2024-09-23 by dkerr64