ratgdo/homekit-ratgdo #228: Connectivity issue with 1.7.0

mgrimace, 2024-08-30:
Hello,
Since around 1.7 my ratGDO has been iconsistent and generally unresponsive to Siri requests and to browsing to the webUI from other devices. 

I have done the usual troubleshooting steps of rebooting the device, and reflashing via USB. I also restarted my home hub (appleTV 4k).

I plugged in my laptop and captured logs from the device directly. Interestingly, when I plugged in the laptop I was able to connect to the webUI and open the door via Siri, but then immediately after opening, Siri did not respond to requests to close the door saying 'garage door unresponsive'.

The logs did nothing further as I was asking Siri to close the door, so I copied everything. Just note that after opening I asked 'what is the garage door status' and 'close the garage door' multiple times.

Logs from a USB connection to device:
https://pastebin.com/YTe3xt17

jgstroud, 2024-08-30:
Just scanning the logs, I see a HomeKit connection but the connection closed before the ratgdo could respond.  Not sure why this would suddenly pop up in 1.7.  We might need a packet capture.  Maybe a yield and a keep alive in the client connection process.  We do that during initial pairing to avoid this.

mgrimace, 2024-08-30:
> Just scanning the logs, I see a HomeKit connection but the connection closed before the ratgdo could respond. Not sure why this would suddenly pop up in 1.7. We might need a packet capture. Maybe a [...]

FWIW it was also happening in the previous release (1.6 I believe) but not in releases before that. 

I personally don’t know how to do a packet capture but I’d be happy to try with instruction!

mgrimace, 2024-08-30:
I can no longer access the webUI and it appears the ratGDO is not connected in my router app.

jgstroud, 2024-09-01:
Does a power cycle bring it back?  It's too bad you don't have the serial log from when it disconnected from the wifi.  Maybe try downgrading to the last release that you know worked and see if it becomes stable again.

mgrimace, 2024-09-05:
> Does a power cycle bring it back? It's too bad you don't have the serial log from when it disconnected from the wifi. Maybe try downgrading to the last release that you know worked and see if it becomes stable again.

Ok, it looks as though my Wyze Doorbell camera is the culprit. Using docker-wyze-bridge, or scrypted, or any service that attempts to stream the video as an RTSP feed into homekit was causing havok in my network environment. For some reason that resulted in ratgdo losing connectivity, which appears more networking than any sort of bug. I'll close this issue, but I'm happy to test further if there's anything useful that could come of this.

closed 2024-09-05 by mgrimace

jgstroud, 2024-09-14:
Maybe you can move the camera and scrypted service to their own vlan?

mgrimace, 2024-09-15:
> Maybe you can move the camera and scrypted service to their own vlan?

Thank you! I ended up replacing the camera and the issue with the ratGDO is completely resolved. I’m not sure what the camera was broadcasting or doing to the network (2.4 ghz wifi in particular), but clearly it was interfering enough to knock out the ratgdo.

jgstroud, 2024-09-15:
The ratgdo is just a very wimpy microprocessor with very limited memory.  It really doesn't take much to cause a denial of service and unfortunately there isn't a lot we can do about that.