ratgdo/homekit-ratgdo #229: Possible to make garage door not a security device

rsmith4321, 2024-09-01:
Would there be any way to make homekit recognize the garage door as a insecure device like a outlet switch? The issue is I have an automation setup where when I'm connected to my cars Bluetooth and approach the house it opens the garage, but because the garage door is a security device it only works if the phone is unlocked and with a confirmation each time. I understand there are ways around this like setting up an unused smart outlet, and having the smart outlet activate the garage door opening. But I was wondering if an option could be made so HomeKit recognizes the garage door like an outlet with lower security level. I assume it the code somewhere it must tell homekit that it's a security device.

jgstroud, 2024-09-01:
The code just tells HomeKit that this is a garage door.  You can either modify the code to register it as a switch or you could use the home bridge plugin with the esphome firmware, which I think has this feature.

We don't have any plans to implement this.

closed 2024-09-01 by jgstroud

rsmith4321, 2024-09-01:
Thanks for the quick response. Sadly I wouldn't know enough to modify this. I know there are a lot of other people with the same issue and some are buying smart plugs just to get around the issue. If you could give an example of modifying the code or even better a toggle to make the option on the settings page that would be really cool.

dkerr64, 2024-09-02:
@rsmith4321 You could try installing the [homebridge-dummy](https://github.com/nfarina/homebridge-dummy) plugin, create a dummy switch.  Then create an automation that opens/closes the door when the dummy switch is turned on/off.  I have not tried this don't know if it will work.

Changing our plugin to pretend that it is a regular switch instead of garage door is not simple and not wise.  They are completely different accessory types in HomeKit, garage door has additional states that a switch or outlet does not have.  As @jgstroud states, we will not do this.

rsmith4321, 2024-09-02:
Thanks for the help, I just didn't know if it was possible. I'm just
getting a raspberry pi and setting up home assistant and going that route I
guess. It's a shame Apple can't just let us do what we want.

jgstroud, 2024-09-02:
I think the homebridge plugin already has this feature if you just switch to the esphome firmware.