ratgdo/homekit-ratgdo #230: Add optional beeping for unattended close

amoiseiev, 2024-09-04:
Hi,

I’m wondering if it’s possible to add a checkbox one can use with a delayed close to make it beep? Basically, looking to replicate myQ behavior for unattended door close - lights and beeps.

I understand that some may not want it, so making it optional will make both camps happy.

dkerr64, 2024-09-04:
Hi @amoiseiev.  I would love to do this but we have not figured out how.  The delayed close feature is implemented differently by myQ and the protocol has not been figured out, at least not when we last looked into this several months ago.

The beeping is done as part of the garage door's built-in close delay.  Because we don't know how that works, we implement our own delay and "manually" flash the lights for the duration of the delay.  But there is no way (that we know of) to tell the door opener to beep.

amoiseiev, 2024-09-06:
Got, so since it's not something existing physical accessories can trigger by transmitting the signal over the wire, it makes it complicated to figure out what the command should be (if it exists in Security 2.0)

dkerr64, 2024-09-06:
It can obviously been done, but we don't know how.  The HA firmware has the same [problem](https://github.com/ratgdo/esphome-ratgdo/issues/42).

I'll mark this as enhancement and keep it open for now... but a quick fix is unlikely.

deepakkoirala, 2025-02-02:
I believe integrating a small piezo buzzer directly into the RaTgo logic board would be a valuable enhancement. The buzzer could be controlled by a digital output pin on the ESP board, allowing it to beep in specific scenarios (e.g., garage door opening, warnings, or alerts).

Key Benefits:

- Enhanced Safety – Provides an audible alert to notify users when the garage door is opening, reducing risks of accidental closures.
- Low Cost, High Value – The buzzer costs less than $1 but adds significant value to the product.
- Standalone Feature – No need to rely on an external smart relay or Home Assistant automation to trigger a buzzer.
- Better Marketability – Additional safety features can serve as a strong selling point beyond the current features.
- Minimal Power Consumption – A small passive buzzer (e.g., 5V or 3.3V) draws very little current, making it an efficient add-on.

Implementation Considerations:
	•	The buzzer can be activated via a GPIO pin on the ESP board.
	•	Can be toggled via firmware settings or an optional jumper for users who prefer silent operation.
	•	Requires a minor PCB redesign, but the added functionality justifies the change.

This feature would significantly enhance user confidence and safety, ensuring a reliable warning system without requiring extra integrations. I’d love to see this implemented, as it directly addresses a safety concern while providing a practical and marketable improvement.

Would love to hear your thoughts on this!

PaulWieland, 2025-02-02:
https://ratcloud.llc/products/ratgdo32-disco

closed 2025-02-02 by PaulWieland