ratgdo/homekit-ratgdo #233: Door Unlocked Notification

djducat, 2024-09-11:
Twice now, Homekit has reported that my garage door was unlocked.  I have no automation around locking and unlocking the door, it's not a feature I even use.  I am not getting any "locked" notifications, just "unlocked".  Is this a bug?

jgstroud, 2024-09-14:
It's probably a bug.  We would need to see the logs when it happens to see what triggered the update.

djducat, 2024-09-14:
next time it happens, I'll grab the logs.  is there a URL for the log?

dkerr64, 2024-09-15:
There is a [script file](https://github.com/ratgdo/homekit-ratgdo/blob/main/viewlog.sh) that you can run on Linux or Mac that will dump the current log and remain connected and report new messages as they occur.  However you would have to leave this running to have any chance of catching the logs as we have very limited memory in which to save past messages.

dkerr64, 2024-10-26:
Believed to be fixed in release 1.8.0

closed 2024-10-26 by dkerr64