ratgdo/homekit-ratgdo #234: Change WiFi Network Feature Request

infamous-pattern, 2024-09-18:
Would it be possible to add a "Change WiFi Network" button? Essentially, do discovery for new WiFi networks, choose the network name and input the password, and then reboot to apply the settings. 

I'm sure it doesn't happen often, but it's a bit of a pain to take down the RATGDO when you need to change your WiFi network.

dkerr64, 2024-09-19:
Yes, known requirement.  Our reservation is that it is potentially harmful... in that if you mistype something you could be left with a device that is not accessible.  Of course the recover for that is USB flash which I suppose is no worse than the current situation.

dkerr64, 2024-09-19:
Closing as duplicate of #224

closed 2024-09-19 by dkerr64