ratgdo/homekit-ratgdo #235: Move dangerous buttons to password protected area

JKaniarz, 2024-09-18:
I'd like to see the Firmware Update, Un-pair HomeKit, and Reboot buttons moved to the password protected page. Those are "admin" level features that could be used to covertly break into the house. (the other buttons are fine because they generate notifications)

dkerr64, 2024-09-19:
Firmware updates and re-pairing are already password protected... if you are not seeing the prompt then it is because browsers cache credentials and only request userid/password if credentials expire or change.  Those and reboot are also wrapped with a "are you sure" popup.

Reboot is not password protected.  This was a conscious decision as reboot is non-destructive, it takes ~30 seconds to reboot and re-connect to the Apple Home network.

JKaniarz, 2024-09-19:
I'm glad to hear it's secure, but it's not obvious that it is. I knew to test in incognito mode, but I wasn't expecting the password prompt to be after the point of no return. I assumed that "are you sure" meant there were no further guardrails. 

Will you consider showing the password prompt right after the first click for the firmware and unpair buttons to make it more obvious?

dkerr64, 2024-09-19:
Yes, let me check that, we can change to do the credentials check before the are you sure pop up. 

Thanks for the suggestion.

dkerr64, 2024-09-19:
I have made this change in PR #227 so will be included in next release.

dkerr64, 2024-09-23:
fixed in v1.7.1

closed 2024-09-23 by dkerr64