ratgdo/homekit-ratgdo #236: ratgdo w/HomeKit firmware inaccessibly for long period

PacificState, 2024-09-21:
I just set up and installed my ratgdo after flashing the latest version of the HomeKit firmware, but am finding it unresponsive for long periods - both via the Home app on my phone, and the browser. I can very intermittently reach it, but the periods it can be accessed in are incredibly short.

It's right next to my mesh router, so there's no question of signal. How can I debug further? Thanks!

dkerr64, 2024-09-21:
There have been similar reports, e.g. #228 which have turned out to be other devices on the same network.  We suspect, but do not know for sure, that Multicast DNS may be the culprit.  HomeKit uses this and some devices may send a lot of MDNS packets.  Our ESP8266 is very small and can get easily flooded with requests.

As I said, we do not know this for sure, but if you have the capability to isolate the ratgdo on its own VLAN then test whether you can access its web page / ping reliably then that would confirm it is something else on the same network.  Note that you may not be able to pair it to your Apple Home when on its own VLAN... unless your router/gateway is configured to reflect MDNS packets across LANs.

dkerr64, 2024-10-06:
Can you try changing the position of the ratgdo.  If you currently have it very close to the garage door opener, can you try moving it further away... for example just let it dangle by the wire connections temporarily to see if the web interface becomes more responsive.  Testing a theory that the GDO may be emitting some RF interference.  Thanks.

PacificState, 2024-10-09:
Hmmm. Yeah, it was attached right on the opener body before via adhesive PCB standoffs - and whilst I admittedly updated to 1.7.1 at the same time, everything now seems to be fine after detaching the board and leaving it dangling.

dkerr64, 2024-10-12:
ok, thank you.  It's unfortunate that letting it dangle seems to fix the problem... not an ideal long term solution, but if it is indeed interference then there is not much we can do.

mdjhnson, 2024-10-13:
I feel silly saying this but maybe wrap it in aluminum foil to reduce the interference.

jgstroud, 2024-10-14:
don't wrap your ratgdo in foil.  you can try just changing horizontal to vertical mounting and see if just the orientation fixes it.

mdjhnson, 2024-10-14:
I completely agree. 

Sorry that was an unhelpful and stupid suggestion on my part. It either wouldn't help or would hurt the Wi-Fi signal more. I blame making the comment after a long road trip with kids.

closed 2024-10-30 by jgstroud