ratgdo/homekit-ratgdo #243: Recover to OTA mode

wing6111, 2024-10-21:
I probably didn't do enough homework and realize my garage opener is a dry contact really late, after I have installed homekit firmware.

I see homekit doesn't support dry contact yet, and the solution is installing ESPHome, and then use homebridge plugin to enable homekit support. However now I'm stuck at how to install ESPHome, and my ratgdo is not OTA anymore, and I have installed homekit firmware.

Is there any suggestion how to recover or reset the ratgdo to OTA mode?

Thanks.

dkerr64, 2024-10-21:
You have two options.

First option is to connect the ratgdo to your laptop with USB cable and install the ESPHome firmware just like you installed the HomeKit version. 

Alternatively (and I have not tested this) you can probably do a firmware update from the HomeKit firmware... select the "Update from local file" option and select the ESPHome firmware binary (which you will need to download first).  This will overwrite the HomeKit firmware and in theory should work.

jgstroud, 2024-10-21:
An even better option would be to write the code for the dry contacts and submit it!

wing6111, 2024-10-22:
Thanks @dkerr64. Option one worked!

Just log a bit details in case anyone got same problem. Initially I posted the issue because I tried to connect ratgdo to my computer and it was not detectable anymore. I thought the reason was that it already has the HomeKit firmware and not OTA mode. I didn't even try to reboot it. My bad.

Then yesterday after I read @dkerr64 response, I directly tried to find the bin file, but no luck, esphome doesn't provide bin file. They do have a way to install esphome firmware with terminal and command. In summary, first you need to install git, python3, then git clone esphome repo, and cd to the directory of esphome repo, and use "esphome run" command to install. 

However, esphome run installed the firmware, but didn't erase HomeKit firmware. After a few other tries, I just went back to ratgdo web interface, and suddenly ratgdo was detectable again, and then I can erase firmware and install esphome firmware through esphome web interface (@dkerr64 first option). And with esphome firmware, ratgdo is functioning and controlling my dry contact garage door opener.

After all the steps, I'm not very sure exactly which step made ratgdo detectable. I don't know why ratgdo was not detectable at the first place either (I have installed the driver and it was detectable and that was how I installed homekit firmware at the first place). Hope those details help someone in the future.

And yes! Very look forward to having homekit version support dry contact in the future!

closed 2024-10-22 by wing6111