ratgdo/homekit-ratgdo #245: allow customization of syslog port

SShah7433, 2024-10-29:
Allow for syslog to be sent to ports other than 514. Many syslog listeners use ports to segment data or listen on non-privileged ports.

jgstroud, 2024-10-30:
How important is that feature to you?  The feature is trivial to implement, but IMHO, seems excessive for this device and use case.  For segmenting data, syslog has different logging facilities, which we use LOCAL0, and it also filters based on hostname.  out of the box rsyslog creates separate logs for every remote host.  

Every setting we add has the downside of chewing up previous bytes of RAM, so I want to understand how important it is.  Also, if we do implement, it must remain UDP.  While technically possible, we definitely will not implement TCP because it has major memory usage implications.

SShah7433, 2024-11-01:
It's nice to have, but it's a low priority. I can work around it fairly easily, so in the spirit of preserving memory for more requested features, let's backlog this.

jgstroud, 2024-11-02:
Thanks for the update.  It's a mostly trivial feature, so we may still add this but we have a few higher priority tasks at the moment.

dkerr64, 2024-11-14:
Added in PR #247 so will be in next release whenever we do that.

phybersplice, 2024-11-15:
Absolutely no rush. 
Thank you.  Very much appreciated whenever it can be implemented.

dkerr64, 2024-11-23:
Implemented in 1.8.2

closed 2024-11-23 by dkerr64