ratgdo/homekit-ratgdo #246: Unable to install 1.8.1 firmware

tycho40, 2024-10-30:
I currently have HomeKit v1.7.1 firmware installed on my ratgdo v2.53i. I saw that 1.8.1 was available, so at first I tried the Update from GitHub option. It spent about a minute trying to update, but the progress never went past 0%. It then rebooted and was still at 1.7.1.

I then downloaded the homekit-ratgdo-v1.8.1.bin file from GitHub and tried the Update from local file option. Same thing -- update doesn't progress, reboots after about a minute, and then still at 1.7.1.

Is there something from the logs I could provide that might help? Thank you!

dkerr64, 2024-10-30:
If you have any logs then yes please provide.

tycho40, 2024-10-30:
Hi there! Thanks for getting back to me. So I tried the Update from GitHub option again, and this time, something different happened. The updating indicator stayed at 00% again for about a minute, and then I think the ratgdo crashed or hung, because the blue light went out and I couldn't bring up the status page. So I had to unplug the power to let the ratgdo reboot.

Here's the Reboot log:

Messages logged immediately before last user requested reboot or reset...

Server upTime:      0 days 00 hrs 13 mins 29 secs

Server uptime (ms): 809145
Firmware version: 1.7.1
Flash CRC: 0xC3762338
Flash length: 625088
Free heap: 17400
Minimum heap: 15712

>> [ 764783] RATGDO: Client 192.168.7.249 requesting: /crashlog (method: HTTP_GET)
>>> [ 764792] RATGDO: Request to display crash log...
>>> [ 764825] RATGDO: Client 192.168.7.249 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [ 764837] RATGDO: SSE Subscription for client 4f2183a4-3ead-40eb-8ec2-658a6ea56b65 with IP 192.168.7.249: event bus location: /rest/events/2, Total subscribed: 3
>>> [ 764855] RATGDO: Client 192.168.7.249 requesting: /showlog (method: HTTP_GET)
>>> [ 764904] RATGDO: Client 192.168.7.249 listening for SSE events on channel 2
>>> [ 767976] RATGDO: Client 192.168.7.249 not listening, remove SSE subscription. Total subscribed: 2
>>> [ 776404] RATGDO: Client 192.168.7.249 requesting: /status.json (method: HTTP_GET)
>>> [ 776484] RATGDO: JSON length: 809
>>> [ 790905] RATGDO: Client 192.168.7.249 not listening, remove SSE subscription. Total subscribed: 1
>>> [ 800992] RATGDO: Client 192.168.7.249 requesting: /index.html (HTTP_GET, type: text/html, length: 5017)
>>> [ 801065] RATGDO: Client 192.168.7.249 requesting: /status.json (method: HTTP_GET)
>>> [ 801136] RATGDO: JSON length: 809
>>> [ 801173] RATGDO: Client 192.168.7.249 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [ 801174] RATGDO: SSE Subscription for client 63382582-e0c7-4408-b307-ac8128788514 with IP 192.168.7.249: event bus location: /rest/events/1, Total subscribed: 2
>>> [ 801188] RATGDO: Client 192.168.7.249 not listening, remove SSE subscription. Total subscribed: 1
>>> [ 801226] RATGDO: Client 192.168.7.249 listening for SSE events on channel 1
>>> [ 804336] RATGDO: Client 192.168.7.249 requesting: /checkflash (method: HTTP_GET)
>>> [ 805329] RATGDO: checkFlashCRC: true
>>> [ 807499] RATGDO: Client 192.168.7.249 requesting: /reboot (method: HTTP_POST)
>>> [ 807500] RATGDO: ... reboot requested
>>> [ 808004] RATGDO: checkFlashCRC: true
>>> [ 809003] RATGDO: WiFi disconnected SSID: xxx, BSSID: xxx, Reason: 8
>>> [ 809125] RATGDO: writing 0 to file rolling

dkerr64, 2024-10-30:
That log shows us things running okay, but version 1.7.1 which for some reason you are unable to upgrade to 1.8.1.

If you could please... before starting the upgrade, open the system log tab.  You can open it in a separate browser window, so have both the log and main web page visible.  The logs should update in real time.  Now try the upgrade and observe the log, copy and paste the relevant section in here.  A successful upgrade log should look something like this...

```
[2024-10-30T17:48:19.742Z] RATGDO: Key: updateUnderway, Value: {"md5":"15dd429c67cd5317775a0f39aadee1a9","size":654016,"uuid":"347a2fd6-5b2f-41b5-a7f1-3a0d68a89721"}
[2024-10-30T17:48:19.752Z] RATGDO: SetGDO Complete
[2024-10-30T17:48:19.777Z] RATGDO: Update: blob
[2024-10-30T17:48:19.781Z] RATGDO: Available space for upload: 1470464
[2024-10-30T17:48:19.784Z] RATGDO: Firmware size: 654016
[2024-10-30T17:48:19.787Z] RATGDO: Flash chip speed 40 MHz
[2024-10-30T17:48:19.796Z] RATGDO: Expected MD5: 15dd429c67cd5317775a0f39aadee1a9
[2024-10-30T17:48:19.839Z] RATGDO: Update progress: 00%
[2024-10-30T17:48:21.004Z] RATGDO: Update progress: 10%
[2024-10-30T17:48:22.199Z] RATGDO: Update progress: 20%
[2024-10-30T17:48:23.431Z] RATGDO: Update progress: 30%
[2024-10-30T17:48:24.628Z] RATGDO: Update progress: 40%
[2024-10-30T17:48:25.854Z] RATGDO: Update progress: 50%
[2024-10-30T17:48:27.101Z] RATGDO: Update progress: 60%
[2024-10-30T17:48:28.329Z] RATGDO: Update progress: 70%
[2024-10-30T17:48:29.561Z] RATGDO: Update progress: 80%
[2024-10-30T17:48:30.799Z] RATGDO: Update progress: 90%
[2024-10-30T17:48:32.032Z] RATGDO: Update progress: 100%
[2024-10-30T17:48:32.076Z] RATGDO: Upload size: 654016
[2024-10-30T17:48:32.083Z] RATGDO: Received MD5: 15dd429c67cd5317775a0f39aadee1a9
[2024-10-30T17:48:32.403Z] RATGDO: Flash MD5: 15dd429c67cd5317775a0f39aadee1a9
[2024-10-30T17:48:32.508Z] RATGDO: Free heap dropped to 15992
[2024-10-30T17:48:32.521Z] RATGDO: Client 192.168.17.237 requesting: /reboot (method: HTTP_POST)
[2024-10-30T17:48:32.525Z] RATGDO: ... reboot requested
```

tycho40, 2024-10-31:
Ok, so I had the system logs running in a separate window. Here's what I get with Update from GitHub:

```
>>> [87256356] RATGDO: Key: updateUnderway, Value: {"md5":"15dd429c67cd5317775a0f39aadee1a9","size":654016,"uuid":"3fac700f-4439-4fb6-acb3-9a8d60c820cd"}
>>> [87256365] RATGDO: SetGDO Complete
>>> [87256427] RATGDO: Update: blob
>>> [87256438] RATGDO: Available space for upload: 1470464
>>> [87256444] RATGDO: Firmware size: 654016
>>> [87256454] RATGDO: Flash chip speed 40 MHz
>>> [87256468] RATGDO: Expected MD5: 15dd429c67cd5317775a0f39aadee1a9
>>> [87256478] RATGDO: Update progress: 00%
>>> [87322689] RATGDO: Update progress: 10%
>>> [87322698] RATGDO: Client 192.168.7.249 not listening, remove SSE subscription. Total subscribed: 1
>>> [87325373] RATGDO: Update progress: 20%
>>> [87387207] RATGDO: Update was aborted
>>> [87387371] RATGDO: Client 192.168.7.249 requesting: /status.json (method: HTTP_GET)
>>> [87387455] RATGDO: JSON length: 812
>>> [87387501] RATGDO: Update: blob
>>> [87387513] RATGDO: Available space for upload: 1470464
>>> [87387526] RATGDO: Firmware size: 654016
>>> [87387534] RATGDO: Flash chip speed 40 MHz
>>> [87387541] RATGDO: Expected MD5: 15dd429c67cd5317775a0f39aadee1a9
>>> [87387549] RATGDO: Update progress: 00%
>>> [87445529] RATGDO: Update progress: 10%
```
then it became unresponsive, the blue light went off, and I had to unplug/replug it back in.

Here's what I get when uploading the `homekit-ratgdo-v1.8.1.bin` file.

```
>>> [ 178188] RATGDO: Key: updateUnderway, Value: {"md5":"15dd429c67cd5317775a0f39aadee1a9","size":654016,"uuid":"c8899a64-1d84-409b-b21c-b9de87d845d1"}
>>> [ 178198] RATGDO: SetGDO Complete
>>> [ 178247] RATGDO: Update: blob
>>> [ 178256] RATGDO: Available space for upload: 1470464
>>> [ 178262] RATGDO: Firmware size: 654016
>>> [ 178268] RATGDO: Flash chip speed 40 MHz
>>> [ 178282] RATGDO: Expected MD5: 15dd429c67cd5317775a0f39aadee1a9
>>> [ 178294] RATGDO: Update progress: 00%
>>> [ 226743] RATGDO: Update progress: 10%
>>> [ 226753] RATGDO: Client 192.168.7.249 not listening, remove SSE subscription. Total subscribed: 1
>>> [ 277415] RATGDO: Update progress: 20%
```
then same as before, it becomes unresponsive, blue light goes out, and I have to unplug it.

Otherwise, everything seems fine under 1.7.1. Any thoughts? Thanks!

dkerr64, 2024-11-01:
What are your pings like?  It looks like WiFi connection is poor... it takes an awful long time between each percentage progress report.

Try moving the ratgdo to slightly different position, even just 6 inches further away from the door opener.

And if that fails then you will need to flash with the USB cable just like the very first install.

vvwwwvv, 2024-11-03:
FWIW, I had the same issue (Update progress hanging) on two different RatGDOs at two different locations.
In once case, the unit became completely unresponsive to web requests and intermittently responsive to pings (more delay than usual). After some time I found the LED on the unit was solid blue. I power cycled and it came back online and completed the firmware update.

Another unit is at my parent's house and I  have been unable to get it update successfully after multiple attempts, and am just leaving it on 1.7.1 for now. It has failed on multiple attempts and required power cycles in order to become responsive again.

Both locations have strong wifi signal.

dkerr64, 2024-11-05:
@vvwwwvv thank you for the report.  We really don't know what is going on here.  I have experienced it once myself, but it fixed itself.  After I got the step ladder out to connect a cable to try and debug it... it magically just worked.  This led me to suspect RF interference.  But we really don't know.

Thanks