ratgdo/homekit-ratgdo #248: Feature Request: Disable LED blinking.

JKaniarz, 2024-11-10:
Could I trouble you for an option to make the LED always on or off? My preference is for always on, but whatever is easiest works. Both current activity settings cause a lot of blinking that's disorienting when the garage is otherwise dark.

jgstroud, 2024-11-11:
You could just put a piece of tape over it

clindst, 2024-11-13:
@JKaniarz Does yours blink constantly like the video I've attached?  Mine does and it drives me crazy even after disabling the "LED on when idle" setting.  Sure, I could put tape over it but it seems like this should be a simple setting. 

I'm not sure this matters but I'm running this on a Sec+ Door protocol.

https://github.com/user-attachments/assets/0ed37545-1ed4-4494-9a00-995574082adf

JKaniarz, 2024-11-14:
I see the same nonstop blinking. Mine is also Sec+.

dkerr64, 2024-11-14:
The LED is supposed to flash when there is activity on the comms link from ratgdo to the door opener.  On my (Sec+ 2.0) it is not unreasonably active.  If I go into my garage and the motion detector senses me, then it gets more active.

However, that said, I am not against adding a control to turn it off.  I'm kind of busy, but will add to to-do for next update.

Thanks.

PaulWieland, 2024-11-14:
The problem is that security + 1.0 is a constant stream of serial communication with the opener, so the led just flashes constantly. To me, the best would be to have the led flash when it receives a network request. This can help verify that network communication is working when trouble shooting.

dkerr64, 2024-11-14:
Added in PR #247 so will be in next release whenever we do that.

dkerr64, 2024-11-23:
Implemented in 1.8.2

closed 2024-11-23 by dkerr64