ratgdo/homekit-ratgdo #252: 1.8.2 - RATGDO Bootloops / unrecoverable if SEC+ is selected for Door Protocol

zacshann, 2024-11-23:
After upgrading from 1.8.1 to 1.8.2 on two v2.53 boards both units went unresponsive after installation, I let them sit overnight and they had not recovered so I went about trying to reflash them. After multiple tests / reinstalls of 1.8.2 it appears if the device is set to SEC+ for the door protocol before upgrading to 1.8.2, or is switched to it after installing 1.8.2 the device boot loops and becomes unresponsive / cannot recover.

Both showed similar symptoms after initial upgrading from 1.8.1 to 1.8.2:

- Unit shows up in client devices on Unifi controller as active
- Units has sporadic pings / timeouts loop (6 responses, 2/3 timeouts, 6 responses, 2/3 timeouts)
- Web Interface unavailable
- Homekit unresponsive

After initial reflash of firmware 1.8.2:

- Unit shows up in client devices on Unifi controller as active
- Unit has consistent pings again, no timeouts
- Web Interface available

Testing 1.8.2:

-- Changing a no reboot required setting like the name works as expected, device still responsive
-- Rebooting the unit with no settings changes works successfully, device still responsive
-- Changing a reboot required setting that isn't the door protocol like Wifi TX power reboots and applies setting successfully, device still responsive
-- Changing Door Protocol to SEC+ from SEC2.0 causes device reboot as expected by device does not recover and goes into boot loop (see attached log file) and experiences same symptoms as the initial upgrade from 1.8.1 to 1.8.2 mentioned above, log files hint to panic/abort and loop sequence. 

If I can provide any additional information or help please let me know.

Log File from one unit in active boot loop after setting SEC+:
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/17879993/esp-web-tools-logs.txt)

dkerr64, 2024-11-23:
Thanks for reporting this.  I will investigate later today.

dkerr64, 2024-11-23:
Fixed in 1.8.3

Thank you for reporting this, and my apologies for introducing this bug.  It highlights the importance of more thoroughly testing before pushing a new release.

Even though I do not have a ratgdo connected to a Sec+ 1.0 door, I was able to reproduce this bug.  It is an odd one, as the "fix" reverts a memory management change made to the HomeKit component that _should_ have nothing to do with which door protocol is selected.

Unfortunately the crash is so early in the boot process that the only way for you to recover will be to flash the new release using the USB cable, just like an original install.  Do not select to erase the device so all your settings are preserved.

Thank you, and again my apologies.

zacshann, 2024-11-23:
Hey David, 

You're very welcome, and no apologies necessary! Its definitely a strange bug that's for sure. I'm just glad my report was helpful. :) 

Thanks for fixing so quickly (and on a weekend), I can confirm after flash 1.8.3 & enabling SEC+ 1.0 everything appears to be working as expected. No worries on preserving settings. I wiped both units a few times to make sure I was able to reproduce the original issue consistently before submitting the bug report, luckily a lot of the QoL updates you've added make reconfiguring these units super easy so its really no big deal. 

Regardless I really appreciate all the effort and time you put into this project, thanks again and have a great rest of your weekend.

hectcastro, 2024-12-07:
Hey @dkerr64,

Apologies for waking up this thread a few weeks after it closed, but I believe I encountered this same issue with my v2.53i after upgrading to version 1.8.2. In an attempt to fix tings after seeing this thread, I flashed version 1.8.3. However, I am still experiencing issues that weren't occurring with 1.8.1:

- Unit shows up in client devices on UniFi controller
- Unit has consistent pings, no timeouts
- Web interface unavailable (requests timeout)

The HTTP server generally isn't responsive, but on some occasions it responds with bits and pieces of the administrative UI, but not all of it. I've attached the logs in case something in there stands out.

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/18049259/esp-web-tools-logs.txt)

dkerr64, 2024-12-09:
@hectcastro this does look very similar, I don't have any ideas right now but I'll look at it more thoroughly when I get time.

jmrobinsgithub, 2025-01-31:
I am running 1.8.4, upgraded via USB and am experiencing almost the exact same issues as [hectcastro]. The web interface does not always respond, when it does, I get a spinning circle most of the time. The device does not normally respond to pings and when it does has 200-300ms response times. I tried flashing with the esphome firmware via USB and the webinterface for that was very respnsive.  It consistently responded to pings with 4ms response. Then I put the homekit firmware back on and same symptoms. It will not allow me to rejoin to my home in homekit (yes, I removed all the device from the home). It's hard to get the logs to load, but when they did, it's full of crashes. The log went up to 255 with this data.

Messages logged immediately before last system crash...
Server upTime:     0 days 00 hrs 00 mins 00 secs
Crash information recovered from EEPROM
Crash # 1 at -1 ms
Restart reason: 255
Exception (255):
epc1=0xffffffff epc2=0xffffffff epc3=0xffffffff excvaddr=0xffffffff depc=0xffffffff
>>>stack>>>

frankea, 2025-06-29:
This Security+ 1.0 bootloop issue has been addressed in PR #273 with comprehensive millis rollover safety fixes and improved Security+ 1.0 initialization handling.

## Root Cause Analysis
The bootloop when switching to Security+ 1.0 appears to be caused by timing arithmetic issues during the wall panel detection and initialization sequence. The symptoms you describe (sporadic pings, timeouts, unresponsive web interface) suggest timing-related problems preventing proper initialization completion.

## Fixes Implemented

### Millis Rollover Safety
PR #273 includes comprehensive fixes for 32-bit timing arithmetic throughout the codebase, specifically addressing timing comparisons in Security+ 1.0 initialization:

```cpp
// Wall panel detection timing (15 second window)
if (currentMillis - serialDetected < 15000)

// Emulation timing (250ms intervals) 
if (emulateWallPanel && (currentMillis - lastRequestMillis) > 250)

// Detection logging (1 second intervals)
if (currentMillis - lastRequestMillis > 1000)
```

### Security+ 1.0 Initialization Improvements
The changes include enhanced timing safety in the Security+ 1.0 setup sequence:
- **Wall panel detection window** with rollover-safe timing prevents infinite detection loops
- **Emulation mode switching** with proper timing prevents initialization hangs
- **Serial communication timing** ensuring proper 1200 baud initialization

## Connection to Your Bootloop
Your specific issue pattern suggests the bootloop occurs when:
1. Device switches to Security+ 1.0 protocol and reboots
2. Wall panel detection timing gets stuck due to rollover edge cases
3. 15-second detection window never properly completes
4. Device can't finish initialization and watchdog resets it

## Technical Details
The critical timing areas that could cause bootloops:
- **15-second wall panel detection window** - if timing arithmetic fails, could loop indefinitely
- **250ms emulation intervals** - improper timing could cause communication failures
- **Serial detection logic** - timing issues could prevent proper mode switching

## Expected Resolution
With these timing fixes, switching to Security+ 1.0 should:
- Complete the 15-second wall panel detection window properly
- Successfully switch to emulation mode if no panel detected
- Maintain stable timing throughout the initialization sequence
- Avoid the bootloop condition caused by timing arithmetic failures

The comprehensive timing safety improvements specifically target the edge cases in Security+ 1.0 initialization that could cause the device to hang during protocol switching.