ratgdo/homekit-ratgdo #255: Homekit Automation shows Lock / Unlock vs Open / Close - 2nd Garage Door RatGDO

spatick1, 2024-12-09:
I have 2 garage doors (Double Door and Single Door) and installed RatGDO on both and added them to Homekit.  I'm adding some automation to turn the lights on / off with the doors opening / closing.  One of the doors is as it should be when I add an automation and the door shows up with open and close.  The other door (screenshot attached) shows only Locks and Unlocks.  I have no locks on either door - just open / close / light.  I have done a HomeKit unpair / repair and that did not change anything.

Lemme know if there's any advice!!

[image]

donavanbecker, 2024-12-09:
This is a HomeKit bug, you will need to setup in a 3rd party app.

spatick1, 2024-12-09:
Thx!  Is there a recommended 3rd party app?  I'm using Homekit for mostly everything now.

donavanbecker, 2024-12-09:
Eve or Controller for HomeKit.

You can make the automation in HomeKit by using one of these apps.

Then when you look at the automation in the Home app you will see it set up under opened/closed.

spatick1, 2024-12-09:
Thank you!  Worked perfectly. PAT

closed 2024-12-14 by dkerr64