ratgdo/homekit-ratgdo #257: Inconsistent HomeKit Connectivity 1.8.3

dfree2350, 2024-12-12:
After 1.8.3 and 18.2 OS my HomeKit will mark each of my door openers as No Response, usually regain connection, then drop again. Device is connected to network as I can reach it via IP.

dkerr64, 2024-12-14:
I have updated my Apple devices to 18.2 and I do not see any problems.  Hopefully what you are seeing is only temporary.

jgstroud, 2024-12-23:
I'm also running this firmware combo and I verified my HomePod is also running 18.2.  Devices going unresponsive usually indicates an mdns problem.  If you run the Discovery app, do you see the ratgdo?  Are your devices on the same subnet / vlan?  Are you running homebridge with BonjourHAP?

dfree2350, 2024-12-23:
I did full resets a couple days ago. and that must’ve gotten it. They’ve been on since then without issues.

closed 2024-12-23 by jgstroud