ratgdo/homekit-ratgdo #259: Dry contact is missing support for obstruction detection

dkerr64, 2024-12-14:
@tlhagan over in issue #254 @Olivierbkk is reporting that dry contact is always showing door as obstructed.  It looks like we don't have support for that yet?

Can you take a look please.

Thanks.

dkerr64, 2024-12-14:
@tlhagan  I have created a new branch at your repo on which to apply any fixes... https://github.com/tlhagan/homekit-ratgdo-drycontact/tree/dry-contact-fixes

tlhagan, 2024-12-14:
Thanks for adding the branch. I did not add any code for obstruction detection because from what I can tell this was already supported. The service timer loop calls obstruction_timer and that all happens in the main loop for all protocol types.

I didn't test this because my GDO has obstruction sensors that have a different voltage, so I haven't even connected them to the ratgdo. But happy to add code if needed.

dkerr64, 2024-12-15:
Yes, you are right.  Thanks.

Olivierbkk, 2024-12-15:
I don't really want to connect the obstruction sensor tbh, as you mention the timer should be good enough.

@tlhagan you can use this if you want to switch from different voltages, I use that for different projects to convert 12VDC to Raspberry Pi and ESP GPIO. Actually I plan to use that also for this project as I have 5V on the gdo control board
https://www.aliexpress.com/item/1005004384499442.html?spm=a2g0o.order_list.0.0.70661802ArUrWN 

[image]

dkerr64, 2024-12-15:
@Olivierbkk The obstruction sensor code in ratgdo is independent of door protocol.. Sec+ or dry contact.  So for that sensor to work you need to wire it.

@PaulWieland is there a way to detect if the obstruction pins are actually wired or not.  If we can detect that, then I can remove the obstruction characteristic from HomeKit if they are not connected?

Olivierbkk, 2024-12-15:
With the D5/D6/D7 GPIO not connected, when I power up the ratgdo it show "obstructed" in apple home. Is that normal? is there any way to remove that alarm?

Olivierbkk, 2024-12-15:
ok so it seems to work, my bad, I didn't understand the way it works. Now I connect D7 permanently to ground and the open and close sensor reading are now correct.

The next I need to know is which GPIO controls opening and closing trigger

dkerr64, 2024-12-15:
> ok so it seems to work, my bad, I didn't understand the way it works. Now I connect D7 permanently to ground and the open and close sensor reading are now correct.
> 
> The next I need to know is which GPIO controls opening and closing trigger

If you are on the ESP8266 then try D1, if on the ESP32 then 17

Olivierbkk, 2024-12-15:
esp8266 for now, I can try esp32 later on
ok i will try D1 now but curious to see how it will control opening and closing with only one GPIO

dkerr64, 2024-12-15:
My understanding is that it is a toggle... so if door is open it acts as a close, and if it is closed it acts as a open.

```
//Dry contact commands (only toggle functionality, open/close/toggle/stop -> toggle)
//Toggle signal
digitalWrite(UART_TX_PIN, HIGH); 
delay(500);
digitalWrite(UART_TX_PIN, LOW);
```

dkerr64, 2024-12-15:
If the door you are trying to control has separate connections for open and close, then we need to modify the code to allow for that.

Olivierbkk, 2024-12-15:
my soldering iron is warming up... I'll let you know shortly

Olivierbkk, 2024-12-15:
my door has two options, a single contact that see to work pretty much as you describe above and the other option is separate connections for open close, let's see if your code works as is first

dkerr64, 2024-12-15:
It's midnight here... so I will not see more until tomorrow.
\

Olivierbkk, 2024-12-15:
noon here... testing now, I will post my results

Olivierbkk, 2024-12-15:
if the closed limit switch is connected I get the trigger when opening in apple home
if the open limit switch is connected I do not get the trigger when opening in apple home, this is probably abnormal

if none of the open or closed limit switch are connected, when I switch in apple home I get the 500ms trigger on the relay.
this may be an issue because with the one button method, triggering while the door is moving actually stops the motion

[image]

Olivierbkk, 2024-12-15:
the way the single button works is this:
initial condition ---> action at the press of the button
if the door is fully closed --->  open until reaching the "open" limit switch or until another button press
if the door is fully open ---> close until reaching the "closed" limit switch or until another button press
if the door is opening ---> stop the motor, door will be partly open, none of the limit switch will be closed 
if the door is closing ---> stop the motor, door will be partly open, none of the limit switch will be closed 
if the door is stopped half way and last motion was closing --->  open until reaching the "open" limit switch or until another button press
if the door is stopped half way and last motion was opening --->  close until reaching the "closed" limit switch or until another button press
It doesn't seem like homekit has the ability to stop the opening or closing motion in the middle; it is a toggle open or close

Olivierbkk, 2024-12-15:
> if the open limit switch is connected I do not get the trigger when opening in apple home, this is probably abnormal

actually I just found out that the trigger happens after 15 seconds which happens to be the time I set in the door close delay
[image]

stupid me... initially I thought the door close delay was the time it needed to close. please ignore that comment

Olivierbkk, 2024-12-15:
On the bench it seemed to work OK so I went ahead and tested on the gate motor. The results were rather different, the relays clicked a few times making the gate going back and forth two or three times before finally moving. Then the gate status was rather hit and miss in the app. I am wondering if in your code you have some input signal debouncing?

Also if there is a way to ignore the "obstruction" signal on the software side it would be one less wire to take care of.

Just for fun here is the state-of-the-art (if you like noodles) testing setup:
[image]

PaulWieland, 2024-12-15:
Deleting as there is nothing missing.