ratgdo/homekit-ratgdo #260: Homekit connections maxing out with Homebridge

csmyrnos, 2024-12-16:
I have the latest Ratgdo hardware and am running Homekit firmware 1.8.3. I’ve noticed that the bottom of the main status page often indicates “HomeKit connected: Yes (8)”. I am also running a Homebridge server on a rpi with several accessories that are tied back into my main Homekit setup. The issue occurs when the Ratgdo disconnects from my router (like during a router update). Since the connections are maxed out, Homekit can’t grab another connection and my garage door becomes non-responsive in Homekit. When I looked at the Ratgdo logs, it looks like it is grabbing multiple client connections from the IP address of my Homebridge server. There are also several log entries indicating “HomeKit: No more room for client connections (max 8)” which tell me it is constantly trying to establish more client connections with Homebridge.

Any idea why Ratgdo is grabbing and hanging onto Homekit client connections from my Homebridge server? Homebridge is a hub that connects to Homekit but I don’t think it is actually a Homekit endpoint itself if that makes sense.

Here are some additional logs:

[2024-12-16T00:54:09.829Z] HomeKit: Got new client 1073694700: local 192.168.4.32:5556, remote 192.168.4.41:52356
[2024-12-16T00:54:09.842Z] HomeKit: Setting Timeout to 500ms
[2024-12-16T00:54:10.524Z] HomeKit: Got new client 1073701692: local 192.168.4.32:5556, remote 192.168.4.57:49327
[2024-12-16T00:54:10.651Z] HomeKit: Setting Timeout to 500ms
[2024-12-16T00:54:10.663Z] HomeKit: [Client 1073701692] Pair Verify Step 1/2
[2024-12-16T00:54:11.300Z] HomeKit: Free heap: 19096
[2024-12-16T00:54:11.376Z] RATGDO: Free heap dropped to 18808
[2024-12-16T00:54:11.423Z] HomeKit: [Client 1073701692] Pair Verify Step 2/2
[2024-12-16T00:54:11.435Z] HomeKit: [Client 1073701692] Found pairing with 6885E891-3FE8-446D-BE9B-8706AB987F3C
[2024-12-16T00:54:11.471Z] HomeKit: [Client 1073701692] Verification successful, secure session established
[2024-12-16T00:54:11.481Z] HomeKit: Free heap: 19344

192.168.4.32 is the Ratgdo
192.168.4.31 is my Homebridge server (running on a rpi)
192.168.4.57 is my AppleTV (the actual Homekit hub)

I do notice the “Pair Verify” steps with the correct client. I don’t see those for the Homebridge IP. Should Ratgo be dropping connections that don’t “pair verify”?

Any help trying to decode and/or troubleshoot this would be appreciated.

dkerr64, 2024-12-16:
I have a Homebridge server on my net as well, but I have never seen it attempt to connect to ratgdo.  I find it odd that homebridge would be attempting that.  Maybe one of your plugins?

I've never seen number of clients connected go above 1 for me... and I have Apple TV, Apple Home Pods, iPads, etc that can all act has a hub... but only one should be active at a time, others are failovers.

Feels like something odd with your Homebridge setup.

dkerr64, 2024-12-16:
Alerting @jgstroud to this issue as he may have more knowledge of HomeKit than I do.

jgstroud, 2024-12-16:
Are you by any chance running the ratgdo plugin in homebridge or anything like that?  I also run homebridge and have never seen it try to connect to my ratgdo, nor do I have any idea why it would ever try to talk to the ratgdo.

csmyrnos, 2024-12-16:
I’m not running the Ratgdo plugin on Homebridge… I currently have the following:

[image]

and

[image]

I also don’t think it is something else running on my rpi because when I reboot Homebridge, all of the connections close and then start adding back up again once Homebridge is up and running.

I’m going to disable my plugins one by one to see if I can find the culprit but would it be possible for Ratgdo to close connections that don’t “pair verify” to prevent stale connections from staying open?

csmyrnos, 2024-12-16:
Alright - So the culprit looks like the Alexa plugin (1st one I tried!)… It is using the HAP to scan for Homebridge instances so it can link all of my Homebridge devices with an Alexa skill. Looks like it was discovering the Ratgdo device but the connection was not cleanly going away when it was not needed. I was able to set a filter on that plugin to only look at my Homebridge IP/port and now it is leaving the Ratgdo alone.

This is a pretty specific case but I do think it would be helpful for the Ratgdo to have some better handling of these types of connections. I appreciate the responses but it looks like my issue is resolved. Thanks!

closed 2024-12-16 by csmyrnos

jgstroud, 2024-12-23:
Glad you figured this out.  We also had a problem with homebridge when using the Bonjour-HAP mdns backend.  Would be nice to fix these problems, but we probably never will.