ratgdo/homekit-ratgdo #261: Not handling millis() rollovers

jgstroud, 2024-12-22:
The code is so stable now that my millis() counter actually rolled rover.  
https://docs.arduino.cc/language-reference/en/functions/time/millis/

My uptime had mysteriously gone back to 0, which originally I thought there must have been a power flicker.   But there was nothing in the logs.  Then today I clicked the wall button and the device went into recovery mode after a single press.  I went back and reviewed my syslog logs and see that the counter reset timeout value was set before the rollover.

jameyers4, 2024-12-25:
I, for one, would appreciate this change incorporated into 1.8.3 at earliest reasonable convenience. I was closing in on 40 days when I had to power down for non RATDGO reasons - I was close (-ish) to the 32 bit counter limit…

jgstroud, 2024-12-27:
PR is available.  There is no way to artificially test this as far as I can tell.  I'm debating whether I should just merge it now or wait 50 days to verify it properly handles rollovers.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.