ratgdo/homekit-ratgdo #265: HomeKit responds but garage door doesn’t operate

4havnfun, 2025-02-16:
About 10% of the time when I command the door to open via CarPlay, CarPlay claims the door is opening but it doesn’t move and I use the Chamberlain opener to get in. My door is using Sec+ 2.00 protocol.

PaulWieland, 2025-02-16:
Most likely issue is your phone having a hard time reaching your homekit hub. When you are connected to the car via carplay, the phone is relying on the cell signal to communicate since the WiFi is used to connect to the car.

Apple's UI will show the door as opening as soon as you press the button, but that isn't a confirmation that your request actually made it to your homekit device.

4havnfun, 2025-02-16:
I’ll try to see if it ever happens when not using CarPlay. I’ll report back either way. Thanks

cowboydaks, 2025-02-28:
I had 1.8.3 installed--constant issues with Homekit on my iphone and Carplay--status was pending and wouldn't update.  I unpaired from Homekit, removed the accessory from Homekit, rebooted everything, and Ratgdo would never complete Homekit pairing.  I downgraded to 1.8.1 which came with my device and was able to immediately connect to homekit and control garage from iphone/Carplay.

xhci, 2025-04-28:
@cowboydaks I have been having issues with homekit (often "No Response") since upgrading to 1.8.3 from the 1.7.x series. It was flawless for weeks at minimum before the upgrade. Now I have constant issues.

xhci, 2025-04-30:
As an update, I _think_ I may have fixed it. I was reading ~-60dBm through the web interface, but added an AP right next to the board.

It seems that the wifi chip on the ESP boards may be more susceptible to interference or overly optimistic with their displayed signal, and that this issue might not be specific to ratdgo. Placing an AP within 3 ft seems to resolve my issue so far. I'll update if things change.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64