ratgdo/homekit-ratgdo #267: Connectivity issue, crash logs, 1.8.3/4

mgrimace, 2025-03-02:
I'm experiencing connectivity issues to the WebUI on both 1.8.3 and 1.8.4.
Here are the crash logs for 1.8.3. After being unresponsive, I manually connected via USB, and updated to 1.8.4, but still having connectivity issues on 1.8.4. Specifically, no or slow response on the webUI.

```
Messages logged immediately before last system crash...

Server upTime:     38 days 08 hrs 18 mins 43 secs

Crash information recovered from EEPROM
Crash # 1 at -981844466 ms
Restart reason: 2
Exception (0):
epc1=0x312f786f epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000b97b depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3fffff60 end: 3fffffb0
3fffff60: 4027688c 3fff0da8 3fff0dd0 60000600 
3fffff70: 65819d07 3fff0dd0 3fff0da8 40276899 
3fffff80: 402768de 3fffdab0 00000000 3fffdcb0 
3fffff90: 3fff0de8 3fffdab0 00000000 00000001 
3fffffa0: 40000f49 40000f49 ffffff01 40000f49 
<<<stack<<<

EEPROM space available: 0x038b bytes

Flash CRC OK
Server uptime (ms): 3313123926
Firmware version: 1.8.3
Flash CRC: 0x50EE20FD
Flash length: 654464
Free heap: 20464
Minimum heap: 17456
IRAM heap size: 18288

 0 to 3
>>> [3311614940] RATGDO: reader completed packet
>>> [3311614941] RATGDO: DECODED  000126A3 000000726698400A 0000E084
>>> [3311614942] RATGDO: PACKET(0x98400A @ 0x126A3) MotorOn - NoData: [Zero: 0x00000000, Parity: 0xE]
>>> [3311614949] RATGDO: Support for MotorOn packet unimplemented. Ignoring.
>>> [3311618982] RATGDO: reader completed packet
>>> [3311618983] RATGDO: DECODED  000126A3 000000726698400A 0000E084
>>> [3311618985] RATGDO: PACKET(0x98400A @ 0x126A3) MotorOn - NoData: [Zero: 0x00000000, Parity: 0xE]
>>> [3311618991] RATGDO: Support for MotorOn packet unimplemented. Ignoring.
>>> [3311622220] RATGDO: reader completed packet
>>> [3311622221] RATGDO: DECODED  000126A3 000000726698400A 0000E084
>>> [3311622222] RATGDO: PACKET(0x98400A @ 0x126A3) MotorOn - NoData: [Zero: 0x00000000, Parity: 0xE]
>>> [3311622229] RATGDO: Support for MotorOn packet unimplemented. Ignoring.
>>> [3311624119] RATGDO: reader completed packet
>>> [3311624120] RATGDO: DECODED  000126A4 000000706698400A 4260B281
>>> [3311624121] RATGDO: PACKET(0x98400A @ 0x126A4) Status - Status: [DoorState Closed, Parity 0xB, Obs 1, Lock 0, Light 1]
>>> [3311624130] RATGDO: tgt 1 curr 1
>>> [3311624134] HomeKit: [Client 1073701764] Got characteristic 1.11 GARAGE_DOOR_OPENER Current Door State change event 1
>>> [3311624144] HomeKit: [Client 1073701764] Got characteristic 1.12 GARAGE_DOOR_OPENER Target Door State change event 1
>>> [3311624266] RATGDO: Current Door State changing from 3 to 1
>>> [3311640955] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3311701654] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3311731248] RATGDO: Gateway 192.168.68.1 alive 186 ms
>>> [3311761691] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3311821696] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3311854158] RATGDO: Client 192.168.71.133 requesting: /index.html (HTTP_HEAD, type: text/html)
>>> [3311854468] RATGDO: Client 192.168.71.133 requesting: /status.json (method: HTTP_GET)
>>> [3311854958] RATGDO: JSON length: 949
>>> [3311882824] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3311927735] RATGDO: reader completed packet
>>> [3311927735] RATGDO: DECODED  000126A5 000000706698400A 40609281
>>> [3311927737] RATGDO: PACKET(0x98400A @ 0x126A5) Status - Status: [DoorState Closed, Parity 0x9, Obs 1, Lock 0, Light 0]
>>> [3311927745] RATGDO: tgt 1 curr 1
>>> [3311927749] RATGDO: Light Status Off
[... 29 lines cut ...]
>>> [3312847004] RATGDO: tgt 1 curr 1
>>> [3312847784] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3312871266] RATGDO: Gateway 192.168.68.1 alive 121 ms
>>> [3312907743] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3312968582] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3312991269] RATGDO: Gateway 192.168.68.1 alive 125 ms
>>> [3313028911] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3313051269] RATGDO: Gateway 192.168.68.1 alive 165 ms
>>> [3313089755] HomeKit: [Client 1073701764] Requested characteristic info for 1.6 ACCESSORY_INFORMATION Firmware Revision 1.8.3
>>> [3313118419] RATGDO: Client 192.168.71.133 requesting: /index.html (HTTP_GET, type: text/html, length: 5273)
>>> [3313120287] RATGDO: Client 192.168.68.83 requesting: /index.html (HTTP_GET, type: text/html, length: 5273)
>>> [3313120535] RATGDO: Client 192.168.68.83 requesting: /style.css (HTTP_GET, type: text/css, length: 2002)
>>> [3313120575] RATGDO: Client 192.168.68.83 requesting: /functions.js (HTTP_GET, type: text/javascript, length: 11651)
>>> [3313121385] RATGDO: Client 192.168.68.83 requesting: /qrcode.svg (HTTP_GET, type: image/svg+xml, length: 2051)
>>> [3313121461] RATGDO: Client 192.168.68.83 requesting: /status.json (method: HTTP_GET)
>>> [3313122021] RATGDO: JSON length: 951
>>> [3313122035] RATGDO: Client 192.168.68.83 requesting: /settings-sliders.svg (HTTP_GET, type: image/svg+xml, length: 355)
>>> [3313122050] RATGDO: Client 192.168.68.83 requesting: /garage-car.svg (HTTP_GET, type: image/svg+xml, length: 621)
>>> [3313122073] RATGDO: Client 192.168.68.83 requesting: /favicon.png (HTTP_GET, type: image/png, length: 943)
>>> [3313122511] RATGDO: Client 192.168.68.83 requesting: /apple-touch-icon.png (HTTP_GET, type: image/png, length: 2721)
>>> [3313122639] RATGDO: Client 192.168.68.83 requesting: /rest/events/subscribe (method: HTTP_GET)
>>> [3313122641] RATGDO: SSE Subscription for client 1ea0f9b1-81fd-40f4-8f67-76fcf5fc42a7 with IP 192.168.68.83: event bus location: /rest/events/0, Total subscribed: 1
>>> [3313122766] HomeKit: [Client 1073701764] Disconnected!
>>> [3313122767] HomeKit: [Client 1073701764] Closing client connection
```

jmrobinsgithub, 2025-03-05:
This sounds very similar to my issues which I have logged in issue 266. Is your homekit connection stable?

mgrimace, 2025-03-06:
> This sounds very similar to my issues which I have logged in issue 266. Is your homekit connection stable?

It's difficult to say, I have not attempted to re-pair it to homekit, and it seems to generate notifications and respond to voice as expected. I just don't open/close the garage that frequently to notice if there are times when its not-responsive. But checking the webUI, it stalls/times out with a 'problem loading page' every time I try and connect. Often fixed after a few refreshes, but not crash to speak of.

mgrimace, 2025-03-06:
> This sounds very similar to my issues which I have logged in issue 266. Is your homekit connection stable?

Correction: no, it is not. I just left (opened/closed the garage) and it did not trigger any notifications as it should.

SShah7433, 2025-05-06:
I had a similar issue, for some reason, it is only stable on specific wifi frequency channels. Channel 9 is the only one I have found that works consistently for me.

mgrimace, 2025-05-06:
> I had a similar issue, for some reason, it is only stable on specific wifi frequency channels. Channel 9 is the only one I have found that works consistently for me.

Thanks! I'll see if I can try that. I do still have unresolved but periodic connectivity issues. Interestingly my server logs shows 27 days server uptime, but my crash logs show 3 days. Here are the latest crashes if it helps any:

```
Messages logged immediately before last system crash...

Server upTime:     3 days 03 hrs 36 mins 54 secs

Crash information recovered from EEPROM
Crash # 1 at -981844466 ms
Restart reason: 2
Exception (0):
epc1=0x312f786f epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000b97b depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3fffff60 end: 3fffffb0
3fffff60: 4027688c 3fff0da8 3fff0dd0 60000600 
3fffff70: 65819d07 3fff0dd0 3fff0da8 40276899 
3fffff80: 402768de 3fffdab0 00000000 3fffdcb0 
3fffff90: 3fff0de8 3fffdab0 00000000 00000001 
3fffffa0: 40000f49 40000f49 ffffff01 40000f49 
<<<stack<<<
Crash # 2 at 272212906 ms
Restart reason: 2
Exception (9):
epc1=0x40255ddf epc2=0x00000000 epc3=0x00000000 excvaddr=0x04f00297 depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3fffff20 end: 3fffffb0
3fffff20: 00001b28 00000365 3ffe8f54 4028af60 
3fffff30: 402534f4 1039a362 3fff35dc 40253504 
3fffff40: 4028af60 00000020 3fff531c 40101588 
3fffff50: 61263a47 1039a362 3fff35dc 40253678 
3fffff60: 4027688c 3fff0da8 3fff0dd0 60000600 
3fffff70: 61263a47 3fff0dd0 3fff0da8 40276899 
3fffff80: 402768de 3fffdab0 00000000 3fffdcb0 
3fffff90: 3fff0de0 3fffdad0 3fff2ebc 4023a06e 
3fffffa0: 40000f49 40000f49 ffffff01 40000f49 
<<<stack<<<
```

frankea, 2025-06-29:
Looking at the crash logs you've provided, this appears to be an Exception (9) crash which typically indicates memory corruption or buffer overflow issues. The discrepancy between your server uptime (27 days) and crash logs (3 days) suggests the device has been restarting more frequently than the web interface indicates.

Our changes in PR #273 specifically address several issues that could be causing these symptoms:

**Web Performance & Connection Issues:**
- Implemented connection throttling to prevent overwhelming the ESP8266's limited resources
- Added JSON response caching to reduce memory allocation during web requests
- Improved buffer management in web handlers to prevent memory corruption

**Memory Safety Improvements:**
- Fixed potential buffer overflows in protocol handling that could cause Exception (9) crashes
- Added bounds checking for all buffer operations
- Improved memory allocation patterns to reduce fragmentation

**Long-term Stability:**
- Enhanced millis() rollover handling to prevent crashes after extended uptime periods
- Improved error handling throughout the codebase to gracefully recover from edge cases

The Exception (9) crash you're seeing (LoadStoreAlignmentCause) combined with web connectivity issues strongly suggests memory corruption, which our buffer safety improvements should resolve. The WiFi channel recommendation is also valuable - some ESP8266 devices are sensitive to specific channels due to hardware variations.

These fixes are currently being tested and should provide better stability for both web interface performance and overall system reliability.

mgrimace, 2025-09-13:
> Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.

Testing 2.0.0 now thanks! First impressions seem stable!