ratgdo/homekit-ratgdo #269: Home Kit-connected ratgdo doesn't appear in Home Assistant

highlamp, 2025-03-23:
Hello!
Love the device, it was super easy to install and configure with HomeKit.
I also use Home Assistant, but it isn't discovering the HomeKit-based ratgdo device for some reason.
Have any ideas on how to troubleshoot?

jameyers4, 2025-03-28:
You’ve not stated which firm where you’ve installed on the device. Did you install the HomeKit firmware or the ESPHome firmware? I do not use home assistant so I can’t comment any further than that. I’m pretty sure to use home assistant you need the ESP home firmware.

highlamp, 2025-03-28:
I am currently using the HomeKit firmware. I was under the impression that home assistant would be able to detect HomeKit-based devices but maybe I was incorrect. I will continue to research on the HA side and report back.

PaulWieland, 2025-03-28:
If you are using home assistant, use the ESP Home firmware.

closed 2025-09-13 by dkerr64