ratgdo/homekit-ratgdo #271: 1.8.3 RATGDO crash - advice/analysis?

jameyers4, 2025-04-19:
I had RATGDO crash the other day after a several months of (as far as I know) solid runs.  Snippet below - please advise.

Messages logged immediately before last system crash...
Server started at: Sat, 29 Mar 2025 09:07:13 GMT
Server crashed at: Wed, 16 Apr 2025 21:59:20 GMT
Server upTime:     18 days 12 hrs 52 mins 06 secs
 Crash information recovered from EEPROM
Crash # 1 at 1601525710 ms
Restart reason: 2
Exception (9):
epc1=0x4022c5af epc2=0x00000000 epc3=0x00000000 excvaddr=0x86bd23b9 depc=0x00000000
>>>stack>>>

ctx: sys

sp: 3fffff10 end: 3fffffb0
3fffff10: 40250000 86bd9cea 000005cc 3fff32f4 
3fffff20: 3fff6c34 3fff6c52 3fff355c 402560ad 
3fffff30: 00000014 00000000 3fff32f4 4010116e 
3fffff40: 40272344 3fff5864 3fff3ba4 3fff4bec 
3fffff50: 3fff32f4 00000008 3fff6c34 4024ea65 
3fffff60: 3fffdc80 3fff3ba4 3fff5864 4024e889 
3fffff70: 40266aed 3fff3ba4 3fff5864 40266afe 
3fffff80: 3fff6c44 3fff6c34 00000000 3fffdab0 
3fffff90: 40271f73 00000000 3fff5864 402764df 
3fffffa0: 40000f49 3fffdab0 ffffff00 40000f49 
<<<stack<<<

EEPROM space available: 0x033b bytes

Flash CRC OK
Server time (secs): 1744840760
Server uptime (ms): 1601526880
Firmware version: 1.8.3
Flash CRC: 0x50EE20FD
Flash length: 654464
Free heap: 21504
Minimum heap: 18032
IRAM heap size: 18288

jameyers4, 2025-05-13:
… anyone?

jgstroud, 2025-05-13:
looks like it crashed somewhere receiving a packet.  Hard to tell much from this.   If it recovered and is otherwise relatively stable, I wouldn't worry too much about it.

dkerr64, 2025-06-28:
@frankea Does the ESP32 have the same limitation?  ie, do I need to look for alignment issues in my structs over there?

@jgstroud you good to merge the PR?

jameyers4, 2025-06-28:
Hallelujah….

frankea, 2025-06-29:
@dkerr64 Good question\! From what I understand, the ESP32 is more forgiving with alignment than the ESP8266. The ESP32's Xtensa LX6 core can generally handle unaligned access (though with a performance hit), while the ESP8266's LX106 is much stricter and will throw those LoadStoreAlignmentCause exceptions we've been seeing.

That said, I don't actually have ESP32 hardware to test this on, so I can't say definitively. But in theory, adding the `__attribute__((aligned(4)))` to the structs on ESP32 would still be beneficial for:
- Better performance (aligned access is faster)
- Code consistency between platforms
- Just being safe in general

The alignment fix is pretty lightweight - just adding the attribute to a couple structs - so it probably makes sense to apply it to ESP32 as well even if it's not strictly required.

Has anyone with ESP32 hardware seen similar crash issues, or has it been running more stable?

jameyers4, 2025-06-30:
Thanks @frankea for the analysis and pending fix.  This one's vexed me a couple of time; I also look forward to the millis rollover fix - the two of these should take the device reliability to the next level.  I appreciate your and everyone else's work to make and improve this useful device.  Looking forward to a release soon.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.