ratgdo/homekit-ratgdo #288: Troubleshooting random open/closing events

chriscarpenter12, 2025-09-15:
I've been on 1.8.3 for quite a while, but have recently been experiencing random open/close events with my garage door. Sometimes multiple times a day it will seemingly open or close on its own. 3 time while I was away it opened from a completely closed state. A lot of times while I'm home I'll crack the garage door open to just vent the air and get circulation inside the garage. I've seen the recent issues (all closed now) about crashes or reboots, but I don't seem to have any problem related to this. 

Steps taken:
- I've enabled syslog forwarding and reviewed the HomeKit security events but they're not really helpful beyond showing an event time.
- Verified no automations are set to open/close the garage door
- Verified my spouse or I didn't purposefully or accidentally trigger the garage door.
- Checked system, reboot and crash logs
- Upgrade to latest 1.9.1 firmware

[image]

Apart from the recent random events everything has worked perfectly. Obstruction sensors, lights and motion has always worked since installed in November 2023.

```
Server started at:   Tue, 09 Sep 2025 23:35:50 GMT
Server current time: Mon, 15 Sep 2025 17:12:37 GMT
Server upTime:       5 days 17 hrs 36 mins 46 secs
Server time (secs): 1757956357
Server uptime (ms): 495406163
Firmware version: 1.9.1
Flash CRC: 0x5D3650A0
Flash length: 658064
Free heap: 15760
Minimum heap: 15760
IRAM heap size: 18288
```

[logs_2025-9-15.csv](https://github.com/user-attachments/files/22348255/logs_2025-9-15.csv)

If there's any additional info I can provide or monitor I'm happy to do so to figure out what's happening.

chriscarpenter12, 2025-09-16:
It is but the relay device was removed when I installed the ratgdo in 2023. For my sanity I’ve also double checked the car garage door remotes we have weren’t accidentally being clicked and I’ve taken the batteries out. We use the built-in homelink buttons in our cars.

jameyers4, 2025-09-16:
I’d check to see if the main Chamberian unit is still connected to your WiFi - if you do nothing with anything chamberlain on the Wi-Fi, perhaps administratively disable network access for the MyQ opener itself on your router. This is a long shot, but I’d do this if I were you.Best, Jeff

chriscarpenter12, 2025-09-16:
The device I think I had was an internet gateway which has long been removed. I’m not aware of this model opener having any other WiFi capability. It’s definitely not on my network or any vlans. 

[image]

My next steps were to fully reset the ratgdo or remove the ratgdo altogether to see if symptoms persist.

chriscarpenter12, 2025-09-16:
> so turn on debug level logging ideally to a syslog server

I am already sending logs to syslog server. I'm not seeing an obvious way to enable debug logs. Admin settings screenshot above.

> What we are looking for is whether a HomeKit action caused it, or something else.

That's what I've been trying to figure out myself too. I hesitated to open an issue here or not, but I did because I can't find a cause.

I was away this morning, no one was home and the door fully opened at 8:28am. Home app activity history and syslog attached.

[image]

[logs_2025-9-16.csv](https://github.com/user-attachments/files/22364276/logs_2025-9-16.csv)

So far the only ways to control the door in my setup I know of:

- The controller on the wall wired in
- The car remotes (all batteries removed) not in use
- The Homelink buttons in our two vehicles
- Homelink [bridge kit](https://www.homelinkestore.com/Products/HomeLink-Compatibility-Bridge-Kit.html) installed since ~2015
- External keypad (with code)
- Home app (ratgdo)

chriscarpenter12, 2025-09-19:
Well I may have found the cause by happenstance. I was out front of the house and the garage door closed just as my neighbor two houses down was pulling in. Monitored external cameras for the next day and every random instance lined up with neighbors activity. I’m kind of perplexed how it happened only within the last month or so. I resent the opener and repaired all my physical devices. Will continue to monitor. 

Having said that it would be helpful from a logging perspective if it was able to distinguish between external door events vs HomeKit sources events. (If it already does I may have missed it)

Thanks to everyone who replied!

chriscarpenter12, 2025-09-19:
> Don't forget the combination keypad if you have one!

Yep, got every reset and re-paired. Giving this another day then I'll self close.

I did have a question regarding the log levels. The [readme](https://github.com/ratgdo/homekit-ratgdo#log-level) mentions setting it, but my settings page has no option for that? v1.9.1

[image]

dkerr64, 2025-09-19:
Logging all changed in 2.0.0 and that is when we introduced ability to set log level.