ratgdo/homekit-ratgdo #291: V1.9.1 -> v2.0.2 = No response post flash

iyerusad, 2025-09-21:
RatGDO 2.5i

Updated from V1.9.1 (~31 days up time) -> v2.0.2 = No response post flash. Guessing network is not connecting but idk.

Have had good luck in the past, first no response flash since probably pre-1.0 days.

Haven't had a chance to debug but presumably will attempt a clean USB flash.

jwhitley, 2025-09-21:
Same here. Upgraded OTA from v1.9.1, nothing but a boot loop on two ratgdo devices. 

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4010083c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffaf0 end: 3fffffb0 offset: 0160
3ffffc50:  00000000 001714d7 40240b1c 00000b54  
3ffffc60:  3fff3d54 0000130f 00001388 00001388  
3ffffc70:  0000130f 00001388 00000adb 40240b9f  
3ffffc80:  00000000 00000adb 220c49ba 000c8d5f  
3ffffc90:  3fff56f4 00000020 0000000c 00001388  
3ffffca0:  3ffffcf8 3fff5754 00000000 4022f706  
3ffffcb0:  40249080 ffffffff 3ffe8f00 4010117e  
3ffffcc0:  00000adb 00000015 3ffffdb8 3ffffd60  
3ffffcd0:  3ffffda4 00000020 3fff56f4 3ffea910  
3ffffce0:  00000050 00000000 3fff56cc 40230a80  
3ffffcf0:  3ffffda4 3ffffdc4 40249080 00000000  
3ffffd00:  3ffffda4 00000000 00000000 00000000  
3ffffd10:  00000000 00000000 3ffffda0 4023b665  
3ffffd20:  00000000 00000000 3ffffda0 4023c10c  
3ffffd30:  00000004 3ffffda0 3ffffd6c 402312e9  
3ffffd40:  3ffffe24 00000030 3ffffe44 3ffffe24  
3ffffd50:  00000fe0 000001fc 3ffe8f40 4010117e  
3ffffd60:  00000000 3ffffd00 00000001 3fffff40  
3ffffd70:  00000000 00000020 3fff5694 40101598  
3ffffd80:  00000000 3ffe8ec0 3ffffe44 3fffff40  
3ffffd90:  00000000 3ffe8ec0 00000001 40227e1c  
3ffffda0:  3fff56cc 612d7069 632e6970 0a006d6f  
3ffffdb0:  00010050 65001388 3fff572c 0015001f  
3ffffdc0:  80000000 70747468 00000000 04000000  
3ffffdd0:  00000000 00000000 00000000 00000000  
3ffffde0:  3fff566c 00000000 3fff566c 0011001f  
3ffffdf0:  80000000 00000000 00000000 00000000  
3ffffe00:  ffffffff 00000000 00000000 4022000a  
3ffffe10:  00000000 00000000 00000000 00000000  
3ffffe20:  00000000 40248cd4 00000000 00001388  
3ffffe30:  3fff4be8 00000000 3fff277c 00000000  
3ffffe40:  00000000 00000000 0026002f 00000000  
3ffffe50:  3ffffea0 3fff4be4 4010945c 3fffff40  
3ffffe60:  00000000 3fff1e1c 3fff4be4 40220848  
3ffffe70:  00000008 3fff1e1c 3ffffec4 3fffff40  
3ffffe80:  00000000 3fff1e1c 4010948c 4022f1cc  
3ffffe90:  3fff56dc 3fffc6fc 00000050 4010a77c  
3ffffea0:  3ffffea8 00000008 656d6974 656e6f5a  
3ffffeb0:  50497200 00000000 00000040 4010948c  
3ffffec0:  00000002 00000000 000b000f 00000000  
3ffffed0:  40249080 03321aac 00000000 40249080  
3ffffee0:  03321aac 00000000 40249080 03321aac  
3ffffef0:  40249080 03321aac 0000011f 3fff41b4  
3fffff00:  3fff4be4 3fff56ac 00000ad0 3fff2a46  
3fffff10:  40260000 00000000 00000000 3fff2a50  
3fffff20:  00000001 3fff580c 4010a794 4022f660  
3fffff30:  40249080 b1321aac 40249080 00ffffff  
3fffff40:  40249080 01321aac 3fff4a64 00000100  
3fffff50:  00000050 3fff580c 4010a77c 4022fb0a  
3fffff60:  3fff580c 401026ee 00000000 00000005  
3fffff70:  00000002 40102484 00000002 00000000  
3fffff80:  401024b9 00000002 00000000 3fff1958  
3fffff90:  3fffdc10 00000000 3fff580c 4028176c  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000afa40
~ld
����'�s��'|�l�d`c��|{�$�'��g�l`��{�l�l��
```

jwhitley, 2025-09-21:
Also, I can see an ephemeral connection to WiFi on my network controller, which agrees with my logs pre-crash. That likewise appears to entirely agree in timeline with @iyerusad's logs – the WiFi connection is just started them BOOM we crash.

dkerr64, 2025-09-22:
Well, the good news is that both stack dumps decode into identical trace backs (below).  Now to figure out why. 

The crash is software Watchdog Timeout (WDT) which means that the code is in a tight loop (or in a delay?) for more time than permitted.  Very odd.  The last of my functions being called is `get_auto_timezone()`... which is called as soon as an IP address is obtained and only if a timezone has not been previously set... but curiously, I am not checking whether user has selected _Enable NTP_.  I should probably have done that.

The timezone info is obtained from http://ip-api.com/csv/?fields=timezone 
If you paste that into a browser what do you get?

```
$ python3 ~/.platformio/packages/framework-arduinoespressif8266/tools/decoder.py homekit-ratgdo-v2.0.2.elf  ~/crash-2025-09-21b.txt
Soft WDT reset
Exception (4) - Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register
epc1=0x4010083c: ets_intr_unlock at ??:?
sp: 3ffffaf0 end: 3fffffb0 offset: 0160

ctx: sys
0x40240b1c: __esp_delay at ??:?
0x40240b9f: esp_try_delay(unsigned int, unsigned int, unsigned int) at ??:?
0x4022f706: _ZL14hostByNameImplPKcR9IPAddressj14DNSResolveType$part$0 at ESP8266WiFiGeneric.cpp:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x4010117e: umm_free_core at umm_malloc.cpp:?
0x40230a80: WiFiClient::connect(char const*, unsigned short) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x4023b665: HTTPClient::connect() at ??:?
0x4023c10c: HTTPClient::sendRequest(char const*, unsigned char const*, unsigned int) at ??:?
0x402312e9: WiFiClient::clone() const at ??:?
0x4010117e: umm_free_core at umm_malloc.cpp:?
0x40101598: free at ??:?
0x40227e1c: get_auto_timezone() at ??:?
0x4022000a: userSettings::toFile(Print&) at ??:?
0x40248cd4: EEPROMClass::write(int, unsigned char) at ??:?
0x40220848: userSettings::get(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at ??:?
0x4022f1cc: onGotIP(WiFiEventStationModeGotIP const&) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40260000: freqcal_scan_done at ??:?
0x4022f660: std::_Function_handler<void (_esp_event*), ESP8266WiFiGenericClass::onStationModeGotIP(std::function<void (WiFiEventStationModeGotIP const&)>)::{lambda(_esp_event*)#1}>::_M_invoke(std::_Any_data const&, _esp_event*&&) at ESP8266WiFiGeneric.cpp:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x40249080: EEPROMClass::write(int, unsigned char) at ??:?
0x4022fb0a: ESP8266WiFiGenericClass::_eventCallback(void*) at ??:?
0x401026ee: ppEnqueueRxq at ??:?
0x40102484: ppProcessTxQ at ??:?
0x401024b9: ppProcessTxQ at ??:?
0x4028176c: wifi_set_status_led_output_level at ??:?
```

iyerusad, 2025-09-22:
Thanks, looks like resolved in v2.0.3 (upgrade 1.9.1 -> 2.0.3 appears successful).

dkerr64, 2025-09-22:
> Thanks, looks like resolved in v2.0.3 (upgrade 1.9.1 -> 2.0.3 appears successful).

Good, thanks for confirming.

jwhitley, 2025-09-22:
@dkerr64 Also confirmed working on v2.0.3 release here. Full steps here for posterity, just in case someone else runs into a similar case of the board in a non-browser-updatable state. Instructions for macOS. Other platforms will necessarily differ:

1. Install `esptool`. macOS and Homebrew users can just do`brew install esptool`
2. Download the desired .bin file from the homekit-ratgdo releases page
3. Connect to the ratgdo board with a micro USB cable.
4. `sudo esptool -b 115200 -p /dev/cu.usbserial-110 write-flash 0x000 </path/to/homekit-ratgdo.bin file>`

dkerr64, 2025-09-22:
That is helpful, I may move into the README file.

Also... it would be helpful to have instructions on how to capture the logs / crash dump when the network was not up.  Did you get that from the browser install tool by clicking on the logs button?

jwhitley, 2025-09-22:
> That is helpful, I may move into the README file.

Go for it. I considered submitting it to the Troubleshooting section.. but I realized that I don’t have the right specifics to offer Windows or Linux users. Shouldn’t be hard to bring those together, though?

> Did you get that from the browser install tool by clicking on the logs button?

Precisely. That was fairly helpful. A take-or-leave-tip: When it’s boot looping, the user will need to scroll back a bit to make it to stop scrolling everything away constantly, to be able to copy a useful log segment.